riscvarchive/riscv-gcc #153: relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata section

sathya281, 2019-03-05:
Dear Experts,
I am compiling a small gpio program in riscv64. I am getting a linker error.
Compilation goes well and failure happens at the end of linking.
I am pasting the error below.

Tool information:
Just now I  downloaded the latest from https://github.com/riscv/riscv-tools

 Caveat: Code starts at 0x80000000. 
/home/shaktilab/riscv/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-putchar.o): In function `_putchar_r':
putchar.c:(.text+0x6): relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata section in /home/shaktilab/riscv/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/home/shaktilab/riscv/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status

compilation command snippet:
riscv64-unknown-elf-gcc -w -mcmodel=medany -static -std=gnu99 -fno-builtin-printf 

linker script:

OUTPUT_ARCH( "riscv" )
ENTRY(_start)

/*----------------------------------------------------------------------*/
/* Sections                                                             */
/*----------------------------------------------------------------------*/

SECTIONS
{

  /* text: test code section */
  . = 0x80000000; 
/*  . = 0x20000; */
  .text.init : { *(.text.init) }

  .tohost ALIGN(0x1000) : { *(.tohost) }

  .text : { *(.text) }

  /* data segment */
  .data : { *(.data) }

  .sdata : {
    __global_pointer$ = . + 0x800;
    *(.srodata.cst16) *(.srodata.cst8) *(.srodata.cst4) *(.srodata.cst2) *(.srodata*)
    *(.sdata .sdata.* .gnu.linkonce.s.*)
  }

  /* bss segment */
  .sbss : {
    *(.sbss .sbss.* .gnu.linkonce.sb.*)
    *(.scommon)
  }
  .bss : { *(.bss) }

  /* thread-local data segment */
  .tdata :
  {
    _tls_data = .;
    *(.tdata.begin)
    *(.tdata)
    *(.tdata.end)
  }
  .tbss :
  {
    *(.tbss)
    *(.tbss.end)
  }

  /* End of uninitalized data segement */
  _end = .;
}

jim-wilson, 2019-03-05:
riscv/riscv-tools is broken and may never be fixed.  Don't use it.  Use riscv/riscv-gnu-toolchain directly instead.

You compiled with -mcmodel=medany, but your libraries were compiled with the default -mcmodel=medlow which won't work with a load address of 0x80000000 on a 64-bit target.  If you use riscv/riscv-gnu-toolchain directly, then you can use --with-cmodel=medany to get a toolchain that defaults  to medany cmodel and has libraries compiled for medany.  See the README.md file for more info on building it.

The missing debug info section error is a cascading error that was fixed over a year ago.  That will go away if you use an up to date toolchain.

ilg-ul, 2019-03-05:
> riscv/riscv-tools is broken and may never be fixed

as long as there is no deprecation notice right in front of the README in that repo, people will continue to use it.

jim-wilson, 2019-03-05:
> > riscv/riscv-tools is broken and may never be fixed
> 
> as long as there is no deprecation notice right in front of the README in that repo, people will continue to use it.

I can't fix that.  I can only tell people after the fact that they never should have used it in the first place.  This needs to be handled higher up inside the RISC-V Foundation.

More specifically, riscv-tools is a submodule of rocket-chip, and only changes when a fix is required by rocket-chip.  Changing the compiler will likely break the rocket-chip regression tests, which would require a lot of work on the rocket-chip side, which they have no incentive to do, so the compiler may never be upgraded.  I'm not a hardware engineer working on rocket-chip, so I can't fix this.  The fix needs to come from the rocket-chip folks, and they have no interest in fixing the problem.  One fix for instance might be to rename riscv-tools to rocket-chip-tools, to make it clear that this isn't a generic RISC-V tool set, but rather a very specific toolset for rocket-chip development.  Another fix might be to remove the compiler from riscv-tools and modify rocket-chip to add riscv-gnu-toolchain as a submodule.

sathya281, 2019-03-07:
Hi Jim,
I get your point. I suppose, we need to start maintaining tools locally. This might be a bigger problem down the lane. Thanks for your inputs.

cheers

closed 2019-03-07 by sathya281

prashanth-embedded, 2021-07-20:
Hi jim,
I am able to build riscv-gnu-toolchain on 18.04.  I am working with sifive_s2_n6 core IP. Here design-rtl and design-arty targets are expecting riscv64-unknown-gcc-elf, but from the build, I got  riscv64-unknown-linux-gnu-elf(names are different). i changed the names and tried it led me another bunch of errors. Am I missing anything?  
Thank you

jim-wilson, 2021-07-20:
You can build either an embedded elf target toolchain or a linux target toolchain from the riscv-gnu-toolchain sources.  There are configure options --disable-linux and --enable-linux that control which target toolchain you get when you run "make".  Without the configure options, "make" builds an embedded elf target toolchain and "make linux" builds a linux target toolchain.  You probably typed "make linux" thinking it was necessary on a linux host, but the linux actually means linux target which you don't want.

prashanth-embedded, 2021-07-21:
prasanth@prasanth-VirtualBox:~/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk$ make PROGRAM=hello TARGET=design-arty CONFIGURATION=debug software
mkdir -p /home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/software/hello/debug/
make -C /home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/software/hello hello \
	PORT_DIR= \
	PROGRAM=hello \
	AR=/opt/riscv/bin/riscv64-unknown-elf-ar \
	CC=/opt/riscv/bin/riscv64-unknown-elf-gcc \
	CXX=/opt/riscv/bin/riscv64-unknown-elf-g++ \
	ASFLAGS="-march=rv64imafc_zba_zbb -mabi=lp64f -mcmodel=medany --specs=nano.specs -O0 -g" \
	CCASFLAGS="-march=rv64imafc_zba_zbb -mabi=lp64f -mcmodel=medany -I/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/bsp/design-arty/install/include --specs=nano.specs" \
	CFLAGS="-march=rv64imafc_zba_zbb -mabi=lp64f -mcmodel=medany -ffunction-sections -fdata-sections -I/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/bsp/design-arty/install/include --specs=nano.specs -DMTIME_RATE_HZ_DEF=32768 -O0 -g" \
	CXXFLAGS="-march=rv64imafc_zba_zbb -mabi=lp64f -mcmodel=medany -ffunction-sections -fdata-sections -I/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/bsp/design-arty/install/include --specs=nano.specs -DMTIME_RATE_HZ_DEF=32768 -O0 -g" \
	XCFLAGS="-DMETAL_WAIT_CYCLE=0" \
	LDFLAGS="-Wl,--gc-sections -Wl,-Map,hello.map -nostartfiles -nostdlib -L/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/bsp/design-arty/install/lib/debug/ -T/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/bsp/design-arty/metal.default.lds" \
	LDLIBS="-Wl,--start-group -lc -lgcc -lm -lmetal -lmetal-gloss -Wl,--end-group" \
	FREERTOS_METAL_VENV_PATH="/home/prasanth/Desktop/risc/sifive/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/venv"
make[1]: Entering directory '/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/software/hello'
/opt/riscv/bin/riscv64-unknown-elf-gcc -march=rv64imafc_zba_zbb -mabi=lp64f -mcmodel=medany -ffunction-sections -fdata-sections -I/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/bsp/design-arty/install/include --specs=nano.specs -DMTIME_RATE_HZ_DEF=32768 -O0 -g  -Wl,--gc-sections -Wl,-Map,hello.map -nostartfiles -nostdlib -L/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/bsp/design-arty/install/lib/debug/ -T/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/bsp/design-arty/metal.default.lds  hello.c  -Wl,--start-group -lc -lgcc -lm -lmetal -lmetal-gloss -Wl,--end-group -o hello
Assembler messages:
Error: -march=rv64imafc_zba_zbb: unknown z ISA extension `zba'
/tmp/cc8mOVeU.s:3: Error: -march=rv64i2p0_m2p0_a2p0_f2p0_c2p0_zba_zbb: unknown z ISA extension `zba'
<builtin>: recipe for target 'hello' failed
make[1]: *** [hello] Error 1
make[1]: Leaving directory '/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/software/hello'
scripts/standalone.mk:278: recipe for target '/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/software/hello/debug/hello.elf' failed
make: *** [/home/prasanth/Desktop/temp/SiFive_S2_n6/sifive_rt2_s21/freedom-e-sdk/software/hello/debug/hello.elf] Error 2

jim-wilson, 2021-07-21:
Already asked and answered in the forums.  Please only report problems in one place.
[https://forums.sifive.com/t/mismatch-in-file-structure-of-riscv-gnu-compiler-toolchain/4990/4](url)
SiFive specific questions don't belong here so the forums are a better place for that, and our customer portal is an even better place if you are a coporate customer.

WindowsXp-Beta, 2022-12-25:
> You compiled with -mcmodel=medany, but your libraries were compiled with the default -mcmodel=medlow which won't work with a load address of 0x80000000 on a 64-bit target. If you use [...]

I encountered the same problem. I've re-configured the `newlib` with `-with-cmodel=medany` and re-compiled it. However, it still says `relocation truncated to fit: ***`. The compile command is 
```
riscv64-unknown-elf-gcc -Wl,-mcmodel=medany -Wl,-melf64lriscv -nostartfiles -Wl,-T/home/ubuntu/OH-3.0.5/device/qemu/riscv64_virt/Liteos.ld -mabi=lp64 -static -fno-common -fno-stack-protector -ffunction-sections -fdata-sections -fno-omit-frame-pointer -Wno-format-truncation -fno-builtin -D__RISC_V__ -D__LITEOS_M__ -g -march=rv64imac -L.  -Wl,--start-group libs/libboard.a libs/libdemo.a libs/libkernel.a libs/libarch.a libs/libutils.a libs/libsec_static.a libs/libcpup.a libs/libbacktrace.a libs/libpm.a libs/libtest_init.a libs/libtest_event.a libs/libtest_hwi.a libs/libtest_mem.a libs/libtest_mux.a libs/libtest_queue.a libs/libtest_sem.a libs/libtest_swtmr.a libs/libtest_task.a libs/libtest_pm.a -lgcc -lnosys -lc -Wl,--end-group -o ./unstripped/bin/liteos  && riscv64-unknown-elf-strip --strip-unneeded "./unstripped/bin/liteos" -o "./bin/liteos"
```
And error info is:
```
[OHOS ERROR] libs/libarch.a(libarch.los_interrupt.o): in function `HalIsIntActive':
[OHOS ERROR] /home/ubuntu/OH-3.0.5/out/riscv64_virt/qemu_riscv64_mini_system_demo/../../../kernel/liteos_m/kernel/arch/risc-v/riscv32/gcc/los_interrupt.c:149:(.text.HalIsIntActive+0x8): relocation truncated to fit: R_RISCV_HI20 against symbol `__stack_chk_guard' defined in .sbss section in /opt/new_gcc_riscv64/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-stack_protector.o)
```

Thanks for help and I can provide any other details if you need.