riscvarchive/riscv-gcc #154: Excess stack read and write instructions in ilp32e

quxianmiao, 2019-03-07:
A simple example below:
```
extern void test (int , double, int, double);
int a;
void fff(double b)
 {
   test (a, b, a, b);
 }
```
Compile it with option "-S -march=rv32em -mabi=ilp32e -O2 -fno-schedule-insns -fno-schedule-insns2", the generated compilation is as follows:
```
fff:
        addi    sp,sp,-28
        sw      ra,24(sp)
        sw      a0,20(sp)
        sw      a1,16(sp)
        lui     a5,%hi(a)
        lw      t1,%lo(a)(a5)
        sw      a2,8(sp)
        sw      a3,12(sp)
        sw      a2,0(sp)
        lw      a5,12(sp)
        sw      a5,4(sp)
        mv      a5,t1
        mv      a1,a3
        mv      a3,a2
        mv      a4,a1
        mv      a2,t1
        call    test
        lw      ra,24(sp)
        lw      a0,20(sp)
        lw      a1,16(sp)
        addi    sp,sp,28
        jr      ra
```
among them, the code
```
        sw      a2,8(sp)
        sw      a3,12(sp)
        sw      a2,0(sp)
        lw      a5,12(sp)
        sw      a5,4(sp)
```
are redundancy and can replaced by
```
        sw      a2,0(sp)
        sw      a3,4(sp)
```
in my opinion.

I simply tracked this problem, these instructions are emitted by function 'emit_push_insn' in expr.c
```
  if (mode == BLKmode
      || (STRICT_ALIGNMENT && align < GET_MODE_ALIGNMENT (mode)))
    {
      /* Copy a block into the stack, entirely or partially.  */

      rtx temp;
      int used;
      int offset;
      int skip;

      offset = partial % (PARM_BOUNDARY / BITS_PER_UNIT);
      used = partial - offset;

      if (mode != BLKmode)
	{
	  /* A value is to be stored in an insufficiently aligned
	     stack slot; copy via a suitably aligned slot if
	     necessary.  */
	  size = gen_int_mode (GET_MODE_SIZE (mode), Pmode);
	  if (!MEM_P (xinner))
	    {
	      temp = assign_temp (type, 1, 1);
	      emit_move_insn (temp, xinner);
	      xinner = temp;
	    }
	}
```
And if I change the 'BIGGEST_ALIGNMENT' to 32, the problem has disappeared.
So I'm curious why are these instructions generated and how to solve this problem.

kito-cheng, 2019-03-14:
I can't reproduce the code gen, the code seems like generated by `-O0`?
```
[kito@iCan riscv-workspace]$ cat t.c
extern void test (int , double, int, double);
int a;
void fff(double b)
 {
   test (a, b, a, b);
 }
[kito@iCan riscv-workspace]$ riscv32-unknown-elf-gcc t.c -o -  -march=rv32em -mabi=ilp32e -O2 -fno-schedule-insns -fno-schedule-insns2 -S
        .file   "t.c"
        .option nopic
        .text
        .align  2
        .globl  fff
        .type   fff, @function
fff:
        mv      a4,a0
        mv      a5,a1
        lui     a3,%hi(a)
        lw      a3,%lo(a)(a3)
        mv      a1,a0
        mv      a2,a5
        mv      a0,a3
        tail    test
        .size   fff, .-fff
        .comm   a,4,4
        .ident  "GCC: (GNU) 8.2.0"
```

jim-wilson, 2019-03-14:
I can't reproduce this either.  Maybe you have a buggy version of gcc?  There have been a few patches that changed these macros, and this was accidentally broken for a couple of days once.

The code in emit_push_insn is trying to align a stack slot that isn't aligned.  This may mean that we have to allocate a second stack slot for something if the first stack slot is unaligned.  This may also be an issue if values are passed unaligned in registers, and the registers are pushed as a block to the stack, which then results in the value being in an unaligned stack slot.  If doubles have 64-bit alignment and the stack has only 32-bit alignment, then every double on the stack is going to be unaligned, and this could cause trouble.  WIth 32-bit stack alignment, we really can't align anything to greater than 32-bits, and this should already be handled correctly.

quxianmiao, 2019-03-21:
@kito-cheng @jim-wilson Sorry, I ignored the changes to my local code. I updated the GCC code to the latest and modified the example, this problem still exists.
The modified code is:
```
extern void test (int , double, int, double, int, double);
int a;

void fff(double b)
{
  test (a, b, a, b, a, b);
}
```
Compile it with option "-S -march=rv32em -mabi=ilp32e -O2 -fno-schedule-insns -fno-schedule-insns2", the generated compilation is as follows:
```
        .align  2
        .globl  fff
        .type   fff, @function
fff:
        addi    sp,sp,-32
        sw      ra,28(sp)
        lui     a5,%hi(a)
        lw      a3,%lo(a)(a5)
        sw      a0,12(sp)
        sw      a1,16(sp)
        sw      a0,4(sp)
        lw      a5,16(sp)
        sw      a5,8(sp)
        sw      a3,0(sp)
        mv      a4,a0
        mv      a5,a1
        mv      a1,a0
        mv      a2,a5
        mv      a0,a3
        call    test
        lw      ra,28(sp)
        addi    sp,sp,32
        jr      ra
        .size   fff, .-fff
        .comm   a,4,4
        .ident  "GCC: (GNU) 8.3.0"
```
@jim-wilson As you said, the code seems to allocate a second stack slot for alignment. But I still can't understand the meaning of these following instructions.
```
        sw      a0,12(sp)
        sw      a1,16(sp)
        sw      a0,4(sp)
        lw      a5,16(sp)
        sw      a5,8(sp)
```
They seem to just copy one copy but the results are not aligned.

jim-wilson, 2019-03-21:
There hasn't been any significant work on the current ilp32e ABI, and the plan is to replace it with something better as part of the EABI project, but not clear when that will happen.

The basic problem here is that the stack has 4-byte alignment, but doubles have 8-byte alignment, and you can't give something greater alignment than the stack, unless you have special code to align the stack which we don't have, so the compiler is getting confused and emitting badly optimized code.

Changing the alignment of doubles to 4 would solve the problem, but is an ABI change, which is bad.  This would mean that structures have different layout for ilp32 and ilp32e which is likely to cause trouble.

Another way to solve the problem is to change the stack alignment to 8, which is also an ABI change, but is not as bad, and there is a compiler option for it, -mpreferred-stack-boundary=3.  This requires compiling all code this way, including library code.  The compiler will use a little more stack space, but you will get much better code for this testcase.  This is the solution I would suggest for the EABI.

The third a is passed at 0(sp) which is the final store.

The third b is passed at 4(sp).  That explains the 2 stores at 4(sp) and 8(sp).

The stores to 12(sp) and 16(sp) and the load from 16(sp) are from the compiler getting confused and failing to optimize the code correctly.

quxianmiao, 2019-03-23:
@jim-wilson Thanks. I think it's more about EABI. So I will close this issue

closed 2019-03-23 by quxianmiao