riscvarchive/riscv-gcc #156: Is this repo still active?

marcopeereboom, 2019-04-19:
If it is, what is the difference with riscv-gnu-toolchain?

jim-wilson, 2019-04-20:
riscv-gnu-toolchain includes binutils, gcc, gdb, newlib, and glibc.  Hence riscv-gnu-toolchain is a complete toolchain.  riscv-gcc is just gcc.

This tree was updated in February to gcc-8.3.  The riscv-gcc-8.3.0 branch contains the FSF gcc-8.3 with a lot of patches backported from mainline.  They are mostly new features and optimizations, not bug fixes, so not the kind of stuff one would normally backport to an FSF GCC release branch.

Otherwise, see the riscv-glibc issue #51 answer.  Everything except the 32-bit glibc details also applies here.

marcopeereboom, 2019-04-20:
I see. I thought this repo was a version or two behind upstream Thank you for the clarification.

closed 2019-04-20 by marcopeereboom