riscvarchive/riscv-gcc #157: Upstream issue

sequencer, 2019-05-26:
Hello, I wondering if there is upstream plan for SiFive gcc branch?
[riscv gcc](https://github.com/riscv/riscv-gcc) is stale now. Most of which is upstreamed. 
However `riscv64-unknown-elf` and `riscv-none-embed` is following this repo.
If this repo doesn’t have a upstream plan, riscv toolchain will become stale and stick on the old version of gcc. 
I hope we can maintain a set of patches which can follow the upstream of GNU.

ilg-ul, 2019-05-26:
I think this question applies to the SiFive fork (https://github.com/sifive/riscv-gcc), where new development takes place. Unfortunatelly that repo has no Issues page.

sequencer, 2019-05-26:
Yes, most of the maintainers are same, I hope we can get relative reply here.

sequencer, 2019-05-26:
I found most of patches has been merged, besides clic support and short forward branch optimization for U7. Thanks for upstreaming!
And I wonder it is possible to follow the official GNU branch to avoid the trouble in cherry picking?

jim-wilson, 2019-05-26:
All development work happens in the FSF tree, except for some SiFive specific patches.

We have some local SiFive patches for things like changing the embedded elf multilib list to support all of our customer targets, and building the html/pdf docs that are not built by default.  There is no plan to upstream these as that would not be appropriate.  These are strictly local customizations.

There are also the CLIC patches.  The CLIC patches are deliberately not in the FSF or riscv.org trees because CLIC is not a RISC-V standard.  The policy for now is that the FSF and riscv.org trees will only support official RISC-V Foundation features, and CLIC is still a draft that is in flux.  We want to avoid having FSF releases support unofficial versions of the CLIC spec.  Meanwhile, the SiFive compiler must support unofficial versions of the CLIC spec because we are designing parts that have CLIC support.

Any serious linux distro should be using the FSF sources.  Debian, Fedora, OpenSuse all use purely FSF trees.  There is nothing in the SiFive tree that would benefit any linux distro.  CLIC support is for embedded parts, not linux.

Both U7 optimization patches are upstream now.  One is in gcc 9, and one had to wait for after the gcc 9 release because it depends on an assembler patch not in the last binutils release.

github automatically disables issues/prs for forks, and I don't have admin access for any of the toolchain trees so I can't fix this.  They have to be explicitly enabled in the admin panel after making a fork.

riscv/riscv-gnu-toolchain (and sifive/riscv-gnu-toolchain) are still being maintained.  They are not development trees.  They follow the FSF releases trees, occasionally upgrading to new FSF releases, and occasionally backporting RISC-V specific patches.  This happens as time permits.  We have the latest binutils and gdb versions.  We have not yet upgraded to the latest gcc version, we will probably wait for it to stabilize first.  glibc is still a mess, because the 32-bit patches are not upstream yet, and require user mode qemu work that no one is working on yet.  Newlib is reasonably up to date but an upgrade to get some recent patches would be useful.  Etc.  This tree is useful for building cross compilers, but is not the primary tree for anything.

ilg-ul, 2019-05-26:
The question was strictly related to embedded.

jim-wilson, 2019-05-26:
There is a reference to arch linux above.  Anyways, the answer is the same for embedded, just ignore the linux bits.  Though I'd add that you can also use crosstool-ng to build cross compilers if you don't want to use riscv-gnu-toolchain.

ilg-ul, 2019-05-26:
> There is a reference to arch linux above

yes, @sequencer wants to pack `riscv-none-embed`, which currently follows the SiFive releases, as an Arch package.

sequencer, 2019-05-27:
Yes, I have checked all commits from fsf/8.2.0->sifive/8.2.0, most of which is merged in fsf/9.1.0, only left these 3 [patches](https://github.com/sequencer/pkgbuilds-riscv-none-embed/tree/master/riscv-none-embed-gcc-stage1) unmerged in fsf. 
I patch FSF branch with the unmerged patches from sifve/8.2.0, ignoring the merged one to provide the newest version of gcc of prefix with `riscv-none-embed-` and `riscv64-unknown-elf-`.
I think this can provide a better version control for Linux distribution package management.

jim-wilson, 2019-05-27:
The two gcc CLIC patches won't work without the corresponding binutils CLIC patches, but are harmless by themselves as long as no one uses the CLIC interrupt handler attributes.

The third patch may cause code size and performance regressions without the corresponding binutils optimization patch.  This is why it was upstreamed after the gcc-9 release.  This is potentially harmful even if no one uses the sifive-7-series mtune option.

You should not be cherry picking patches from the SiFive tree unless you know what you are doing.  If you insist on cherry picking gcc patches from us, then you must also cherry pick the related binutils patches.

sequencer, 2019-05-27:
So sorry to mess that up.
Would you mind telling which binutils commit should I pick to provide a suitable binutils with this method?

jim-wilson, 2019-05-27:
If you clone freedom-tools and look at src/riscv-binutils, the top 3 commits are

commit 164267155c96f91472a539ca78ac919993bc5b4e
Merge: 8020f18 162c2e5
Author: Carsten Gosvig <40368726+cgsfv@users.noreply.github.com>
Date:   Fri Apr 5 11:53:56 2019 -0700

    Merge pull request #3 from sifive/sifive-7-integration
    
    RISC-V: Compress 3-operand beq/bne against x0.

commit 162c2e512ece134823685cfcba805b12879893f6
Author: Jim Wilson <jimw@sifive.com>
Date:   Wed Apr 3 11:25:45 2019 -0700

    RISC-V: Compress 3-operand beq/bne against x0.
    
    backport patch from upstream
    
    	gas/
    	* config/tc-riscv.c (validate_riscv_insn) <'C'>: Add 'z' support.
    	(riscv_ip) <'C'>: Add 'z' support.
    	opcodes/
    	* riscv-opc.c (riscv_opcodes) <beq>: Use Cz to compress 3 operand form.
    	<bne>: Likewise.

commit 8020f1860d001b0fd626739fabe0a3504819423b
Author: Jim Wilson <jimw@sifive.com>
Date:   Thu Feb 21 19:21:23 2019 -0800

    Add support for SiFive CLIC CSRs.

sequencer, 2019-05-27:
Thank you so much! I didn't mean to mess that up.
I'll follow your instruction to modify the package scripts to make it clean.

closed 2020-11-09 by sequencer