riscvarchive/riscv-gcc #159: how can I add assemble constraint to constrain operand only use even register?

weiqi7777, 2019-06-17:
hi:
I want to add assemble constraint to constrain operand only use even register.
for example below code
`
asm volatile (
        "add %0, %1, %2 \n"
        : "=&r"(x10)
        : "r"(a), "e"(b)
        :
    );
`
the %2 is constrained to use even register only because of the e constrain descriptor。
how can I achieve it ? 
thanks

aswaterman, 2019-06-17:
if you make `b`'s type `long long` on RV32, or `long __attribute__((mode(TI)))` on RV64), then it will force allocation of an even-odd register pair.  Unfortunately, that might not be exactly what you want.  If you really want to support the `e` constraint, you need to modify GCC to teach it what `e` means.

weiqi7777, 2019-06-17:
@aswaterman  thank for you replay.
can you give me some guide to add e constraint in riscv?
thanks

jim-wilson, 2019-06-17:
You need to add a define_register_constraint to the file gcc/config/riscv/constraints.md for the new e constraint.  It will need to refer to a register class.  See enum reg_class, REG_CLASS_NAMES, REG_CLASS_CONTENTS in the riscv.h file.  You may also need other changes to make this work.

weiqi7777, 2019-06-18:
@jim-wilson  thans for you replay. I will try it.

closed 2019-06-18 by weiqi7777