riscvarchive/riscv-gcc #161: Optimization of logical AND operations on RISC-V 64

fzi-hielscher, 2019-07-05:
We have encountered some odd behavior when compiling for RISC-V 64 with optimizations enabled on recent gcc versions (including 8.3.0). It appears to be related to the following instruction split pattern introduced in commit 50214588c2bcddf1a756861a02543e3f9dc7d52b:

```
(define_split
  [(set (match_operand:GPR 0 "register_operand")
	(and:GPR (match_operand:GPR 1 "register_operand")
		 (match_operand:GPR 2 "p2m1_shift_operand")))]
  ""
 [(set (match_dup 0)
       (ashift:GPR (match_dup 1) (match_dup 2)))
  (set (match_dup 0)
       (lshiftrt:GPR (match_dup 0) (match_dup 2)))]
{
  operands[2] = GEN_INT (BITS_PER_WORD
			 - exact_log2 (INTVAL (operands[2]) + 1));
})
```

Under certain (rare?) conditions the combiner generates logical-AND instructions in SI-Mode even for RV64. Since the "GPR" iterator also matches against SI-Mode registers, this pattern can create shift instruction pairs exceeding the width of the SI-Mode destination register. Consequently, the register will be optimized to be constant zero, even though this does not match the behavior of the original RTX.

An example snippet exposing this problem is:
```

unsigned int test_me(unsigned int foo) {
    unsigned int bar;
    // This (correctly) combines to: bar = foo & 0xFFFFFF;
    bar = foo >> 24;
    bar = bar * (1 << 24);
    bar = foo - bar;
    return bar; 
}
```
Compiling with `riscv64-unknown-elf-gcc -fdump-rtl-combine -march=rv64imafdc -mabi=lp64d -O1 -S` results in the following assembly:

```
	.file	"minifail.c"
	.option nopic
	.text
	.align	1
	.globl	test_me
	.type	test_me, @function
test_me:
	li	a0,0
	ret
	.size	test_me, .-test_me
	.ident	"GCC: (GNU) 8.3.0"

```

In the combiner dump we can see the problematic combine and split:

```
Splitting with gen_split_34
allowing combination of insns 6, 8 and 10
original costs 4 + 4 + 4 = 12
replacement costs 4 + 4 = 8
deferring deletion of insn with uid = 6.
modifying insn i2     8: r78:SI=r75:DI#0<<0x28
      REG_DEAD r75:DI
deferring rescan insn with uid = 8.
modifying insn i3    10: r78:SI=r78:SI 0>>0x28
```

Having very little experience with the internals of gcc, I still attempted to fix this by modifying the original pattern to only match against the target's native mode and adding a new pattern for RV64 which forces the destination and operand to DI-Mode:

```
(define_split
  [(set (match_operand:X 0 "register_operand")
        (and:X (match_operand:X 1 "register_operand")
                 (match_operand:GPR 2 "p2m1_shift_operand")))]
  ""
 [(set (match_dup 0)
       (ashift:X (match_dup 1) (match_dup 2)))
  (set (match_dup 0)
       (lshiftrt:X (match_dup 0) (match_dup 2)))]
{
  operands[2] = GEN_INT (BITS_PER_WORD
                         - exact_log2 (INTVAL (operands[2]) + 1));
})

(define_split
  [(set (match_operand:SI 0 "register_operand")
        (and:SI (match_operand:GPR 1 "register_operand")
                 (match_operand:GPR 2 "p2m1_shift_operand")))]
  "TARGET_64BIT"
 [(set (match_dup 0)
       (ashift:DI (match_dup 1) (match_dup 2)))
  (set (match_dup 0)
       (lshiftrt:DI (match_dup 0) (match_dup 2)))]
{
 operands[2] = GEN_INT (BITS_PER_WORD
                         - exact_log2 (INTVAL (operands[2]) + 1));
  PUT_MODE(operands[0], DImode);
  PUT_MODE(operands[1], DImode);
})

```

The above snippet now is compiled to:
```

test_me:
        slli    a0,a0,40
        srli    a0,a0,40
        ret

```

But I have no idea, whether this is the proper way of doing it.

aswaterman, 2019-07-05:
Thank you for the bug report.  I am suspicious of the BITS_PER_WORD in that pattern; it should depend on wether DImode or SImode is chosen.  I'm testing a fix.

aswaterman, 2019-07-05:
@jim-wilson this patch appears to do the trick:

    diff --git a/gcc/config/riscv/riscv.md b/gcc/config/riscv/riscv.md
    index 668668f..b96e2fe 100644
    --- a/gcc/config/riscv/riscv.md
    +++ b/gcc/config/riscv/riscv.md
    @@ -1759,7 +1759,7 @@
       (set (match_dup 0)
            (lshiftrt:GPR (match_dup 0) (match_dup 2)))]
     {
    -  operands[2] = GEN_INT (BITS_PER_WORD
    +  operands[2] = GEN_INT (GET_MODE_BITSIZE (GET_MODE (operands[2]))
     			 - exact_log2 (INTVAL (operands[2]) + 1));
     })

jim-wilson, 2019-07-07:
Looks reasonable, but I'm concerned why this wasn't caught by testing.  I think we need to add a risc-v specific testcase for this.

aswaterman, 2019-07-07:
I don’t know how people decide what tests should be arch-specific or arch-independent, but my intuition is that a test for this kind of bug should run on all architectures.

jim-wilson, 2019-07-08:
My rule of thumb is that if it is a bug in a target independent optimization then I make it a target independent test, and if it is a bug in a target dependent optimization, then I make it a target dependent test.  This is a target dependent opt as it depends on a riscv specific splitter.  As a target dependent test, I can verify that we get the right riscv instructions.

Another issue is that if I make a target independent test, then this is more work as I have to make the test work for all targets.  As a target dependent test, I only have to make it work for rv32 and rv64.  Maybe this shouldn't be a consideration but I have way too much stuff to do and need to reduce my workload somehow.

I'm adding a testcase, and then will commit upstream.  Since it was reported here I probably need to backport.  I have a list of patches I need to backport and can add it to the list.

jim-wilson, 2019-07-08:
While testing I discovered that the fix isn't quite right, because operands[2] is a VOIDmode CONST_INT, and taking the bitsize from there gets an answer of 0 resulting in shifts with negative counts.  We have to use operands[1] instead to get the right bitsize.

jim-wilson, 2019-07-08:
Fixed upstream
https://gcc.gnu.org/ml/gcc-cvs/2019-07/msg00395.html

fzi-hielscher, 2019-07-08:
Thank you. I totally forgot that RV64 also implements the 32-bit shift operations.

Since it has not been patched in this repository yet, I will keep the issue open for reference. But feel free to close it.

aswaterman, 2019-07-08:
Cool. Thanks.

kito-cheng, 2019-07-15:
I am working on back port patches from gcc 9 and trunk now.

jim-wilson, 2019-07-15:
Thanks for the backports.

It would be nice to get this patch into the FSF gcc-9 branch too.  The linux distros use the FSF branches.  If we add it there, they will eventually automatically get the patch.  Otherwise they would have to add it as a special RISC-V specific patch to their gcc RPM which is inconvenient.

kito-cheng, 2019-07-17:
Back port to `riscv-gcc-8.3.0` branch, and PR created for `riscv-gnu-toolchain`

kito-cheng, 2019-07-18:
Committed to FSF/upstream `gcc-9-branch`.

kito-cheng, 2019-07-21:
Merged into `riscv-gnu-toolchain`

closed 2019-07-21 by kito-cheng