riscvarchive/riscv-gcc #163: How to properly constrain an paired register for inline assembly?

rgwan, 2019-08-01:
Hello everyone:

I want to add inline assembly constraint to constrain an 64-bit register pair on RV32, which uses even register for low 32-bit and low 32-bit register's address + 1 for high 32-bit.
Like the following code:

```
int64_t result;
int32_t a;
int32_t b;
asm volatile ( "muladd64 %0, %1, %2 \n" : "+u"(result) : "r"(a), "r"(b));
```

For example, 'u' modifier will constraint 'result' variable to register pair x10 for low 32-bit and x11 for high 32-bit.

How can I implemented it? I have tried to add SPARC's 'U' modifier to RISC-V port, but it doesn't work.

Thanks!

aswaterman, 2019-08-01:
AFAIK, the only way to do this is to add something equivalent to SPARC's U modifier.  But I think you are on your own to figure out how to make that work.

rgwan, 2019-08-01:
@aswaterman I think it may need some hook on register allocation in RTL pass. But I'm not familiar with gcc. It seems I must to do something with reload to rename pseudo register (low 32-bit) to an proper even register. But I don't really know how to hook up to force the GCC to reload and rename it properly... Could you please give me some advice or guide?

aswaterman, 2019-08-01:
Unfortunately, I'm not an expert on the GCC internals, either.  I learned enough to make the initial RISC-V port by reading the source code of other ports.

Looking at the SPARC source code, it looks like their U modifier is only supported when LRA (the new register allocator) is disabled.  When LRA is enabled, they suppress support for the LDD/STD instructions that use the U modifier.  RISC-V requires LRA, so it seems the SPARC port's approach won't work here.

Some other folks might have more actionable advice.

jim-wilson, 2019-08-01:
The sparc U constraint is an optimization when emitting assembly.  If a double word value happens to end up in an even reg, then we can use a load/store double instruction, otherwise, we split it and use two regular loads/stores.  This isn't what you want.  Unless maybe it is OK to only use the new instruction if values accidentally get the right register alignment.

To force a value to an even register, you need to modify HARD_REGNO_MODE_OK to refuse to allow a double word mode in an odd numbered register in the target register class.  The MIPS target does this for instance.  It has
  if (GP_REG_P (regno) && mode != CCFmode && !MSA_SUPPORTED_MODE_P (mode))
    return ((regno - GP_REG_FIRST) & 1) == 0 || size <= UNITS_PER_WORD;

Unfortunately, if you do this for RISC-V you break the ABI, because the ABI requires unaligned double word values in registers.  Fixing this gets complicated.  You can either define your own ABI which requires aligned double word values in registers.  Or else maybe you can define a new register class that excludes the argument passing registers, and then force double-word values to be aligned in this new reg class, and restrict your new instruction to use this register class.  This will require copying values in and out of arg registers though, which will make the code larger and slower.

rgwan, 2019-08-02:
@jim-wilson 

I'm trying to define a new register class, and force double-word values to be aligned to this new reg class. But in function 'riscv_hard_regno_mode_ok', I don't know how to constraint type or reg_class type. Could you please help me out?

jim-wilson, 2019-08-02:
In riscv_hard_regno_mode_ok you have only mode and register number.  So map the register number to a reg class and then decide if the mode is OK for that reg class.

I don't think that defining a new register class really solves the problem.  If that reg class overlaps the arg passing regs, you still break the ABI.

kito-cheng, 2019-08-02:
I've tried that when implement P extension, restrict DImode can only use even register pair in riscv_hard_regno_mode_ok is work without break ABI magically, but it would cause some silly redundant move for moving argument from odd register pair even register pair in some situation.

rgwan, 2019-08-02:
I think maybe I can add a sub mode for DI and define a constraint modifier. This modifier will change the mode to new sub mode by reload rtx. Then use riscv_hard_regno_ok to decide using even-odd paired register or not.

Maybe it is the better way to solve this problem in SIMD extension.

rgwan, 2019-08-05:
> In riscv_hard_regno_mode_ok you have only mode and register number. So map the register number to a reg class and then decide if the mode is OK for that reg class.
>
> I don't think that defining a new [...]

@jim-wilson  I tried  `if ((regno % 2 != 0) && (mode == E_DImode)) return false;`, It seems it doesn't break the ABI. But I still want to ask you how to properly define a mode that is an alias for DI mode to properly constrain a register pair if you have any spare time.

Sincerely,

Zhiyuan

jim-wilson, 2019-08-07:
Kito commented on the ABI issue.  Apparently the ABI is OK because we hard code register numbers without checking if the mode is OK for that register number.  But if you try to use an arg passed unaligned in regs, then you will get code to copy it to an aligned register first before it gets used, which means less efficient code than is ideal.

Modes directly relate to types.  You can't create a new mode unless you are creating a new kind of type.  Changing modes requires a type conversion (which may be a nop).  Etc.  Modifying riscv_hard_regno_mode_ok is the proper way to constrain a register pair.

GCC doesn't offer any way to align a register pair for one instruction but not for another instruction. 
 Aligning a register pair is done in the register allocator, and either happens always or never depending on how the port is defined.  The base ISA didn't include any register pair support, and the existing ABI didn't consider register pairs when it was designed, so currently we don't do any register pair alignment.

kito-cheng, 2021-02-17:
Long time no further discussion, so I gonna closing this issue.

In case there is anyone still seeking for the solution here, reference implementation:
https://github.com/andestech/gcc/blob/d85979b720a31875155bb742ebafa0801c6b1b85/gcc/config/riscv/riscv.c#L4654

closed 2021-02-17 by kito-cheng