riscvarchive/riscv-gcc #164: Have multilib-generator support mcmodel=medany

galak, 2019-08-07:
For `t-elf-multilib` it would be nice to support that 64-bit multilibs are built with 'mcmodel=medany'.  In some discussion on IRC with @palmer-dabbelt the feeling is that this makes some sense for `t-elf-multilib` but not for `t-linux-multilib`.

So not sure how best to support this, adding a command line option to the script?

Opening this issue for discussion on how to handle this.

palmer-dabbelt, 2019-08-07:
@jim-wilson what do you think about

````
commit 28a52c8c2af3b074ea13a018696af13531794774
gpg: Signature made Wed 07 Aug 2019 11:55:38 AM PDT
gpg:                using RSA key 00CE76D1834960DFCE886DF8EF4CA1502CCBAB41
gpg:                issuer "palmer@dabbelt.com"
gpg: Good signature from "Palmer Dabbelt <palmer@dabbelt.com>" [ultimate]
gpg:                 aka "Palmer Dabbelt <palmer@sifive.com>" [ultimate]
Author: Palmer Dabbelt <palmer@dabbelt.com>
Date:   Wed Aug 7 09:42:16 2019 -0700

    RISC-V: Default to medany on 64-bit ELF targets
    
    Many users of RISC-V toolchains are interested in embedded systems,
    where riscv*-*-elf is the standard tuple.  These systems tend to have
    memory distributed throughout the address space.  For 32-bit systems the
    default medlow code model is fine as it covers the entire address space,
    but for 64-bit systems the medany code model is commonly used because
    there tend to be memories outside the range of medlow.
    
    As a result, pretty much every toolchain vendor is setting some sort of
    CFLAGS_FOR_TARGET="... -mcmodel=medany ..." when building their
    toolchians.  This causes two issues:
    
    * The vendors that forget end up with broken toolchians -- specifically
      this recently came up in the context of the Zephyr SDK.
    * The vendors that remember still end up with medany builds of the
      32-bit libraries.  There's a small performance penalty here, but also
      a possible usability headache related to the PC-relative relocation
      alignment issue.
    
    This patch changes the default code model to medany on riscv*-*-elf
    toolchains when generating code for 64-bit targets.  Another option
    would be to introduce a mechanism into t-elf-multilib (presumably via
    multilib-generator) to allow users to select the code model for the
    64-bit libraries, but this seems like a more user-friendly option to me.
    
    The obvious issue here is that this patch changes the default behavior
    of riscv*-*-elf-gcc, but given the change matches what most users are
    asking for I think it's sane.
    
    2019-08-07  Palmer Dabbelt  <palmer@sifive.com>
    
            * config/riscv/riscv.h (TARGET_DEFAULT_CMODEL): Remove
            definition.
            * config/riscv/elf.h (TARGET_DEFAULT_CMODEL): Moved from
            riscv.h.
            * config/riscv/linux.h (TARGET_DEFAULT_CMODEL): Likewise.
            * config/riscv/riscv-opts.h (enum riscv_cmodel): New CM_DEFAULT
            entry.
            * config/riscv/riscv.c (riscv_option_override): Select a code
            model based on the ISA.

diff --git a/gcc/config/riscv/elf.h b/gcc/config/riscv/elf.h
index 9be97b86629f..6c9f0618bbbc 100644
--- a/gcc/config/riscv/elf.h
+++ b/gcc/config/riscv/elf.h
@@ -32,3 +32,5 @@ along with GCC; see the file COPYING3.  If not see
 
 #undef  ENDFILE_SPEC
 #define ENDFILE_SPEC "crtend%O%s"
+
+#define TARGET_DEFAULT_CMODEL CM_DEFAULT
diff --git a/gcc/config/riscv/linux.h b/gcc/config/riscv/linux.h
index 07ce80a847c9..92bb5972dd32 100644
--- a/gcc/config/riscv/linux.h
+++ b/gcc/config/riscv/linux.h
@@ -74,3 +74,5 @@ along with GCC; see the file COPYING3.  If not see
    "/usr/lib" XLEN_SPEC "/" ABI_SPEC "/ "	\
    "/lib/ "					\
    "/usr/lib/ "
+
+#define TARGET_DEFAULT_CMODEL CM_MEDLOW
diff --git a/gcc/config/riscv/riscv-opts.h b/gcc/config/riscv/riscv-opts.h
index d00fbe20f27d..b2b971703a02 100644
--- a/gcc/config/riscv/riscv-opts.h
+++ b/gcc/config/riscv/riscv-opts.h
@@ -33,6 +33,7 @@ enum riscv_abi_type {
 extern enum riscv_abi_type riscv_abi;
 
 enum riscv_code_model {
+  CM_DEFAULT,
   CM_MEDLOW,
   CM_MEDANY,
   CM_PIC
diff --git a/gcc/config/riscv/riscv.c b/gcc/config/riscv/riscv.c
index 431e90a89572..c95366ef0808 100644
--- a/gcc/config/riscv/riscv.c
+++ b/gcc/config/riscv/riscv.c
@@ -4531,6 +4531,10 @@ riscv_option_override (void)
   /* Function to allocate machine-dependent function status.  */
   init_machine_status = &riscv_init_machine_status;
 
+  /* If the user didn't specify a code model then automatically pick one.  */
+  if (riscv_cmodel == CM_DEFAULT)
+    riscv_cmodel = UNITS_PER_WORD == 4 ? CM_MEDLOW : CM_MEDANY;
+
   if (flag_pic)
     riscv_cmodel = CM_PIC;
 
diff --git a/gcc/config/riscv/riscv.h b/gcc/config/riscv/riscv.h
index 5fc9be8edbf2..ad5332aec633 100644
--- a/gcc/config/riscv/riscv.h
+++ b/gcc/config/riscv/riscv.h
@@ -63,8 +63,6 @@ along with GCC; see the file COPYING3.  If not see
 %{mabi=*} \
 %(subtarget_asm_spec)"
 
-#define TARGET_DEFAULT_CMODEL CM_MEDLOW
-
 #define LOCAL_LABEL_PREFIX	"."
 #define USER_LABEL_PREFIX	""
 
````

jim-wilson, 2019-08-07:
I think this looks reasonable.

Changing the default for elf would reduce the support load a bit, as one of the most common questions for people building their own 64-bit toolchains is asking why they got a relocation overflow, and then I have to tell them to rebuild using --with-cmodel=medany.

I think the comment in riscv.c should be changed.  It says "if user didn't specify a code model", that could perhaps instead say something like "if neither user nor OS port specified a code model" as there are two ways to specify a default code model.

Also, the doc entry in gcc/doc/invoke.texi needs to be updated, as it says that medlow is always the default.  That is not true with this patch.

There are also freebsd and rtems ports.  rtems includes the riscv/elf.h file so is OK.  But freebsd does not include elf.h or linux.h, so it also needs a definition.  Or alternatively, you could leave the TARGET_DEFAULT_CMODEL in riscv.h, and then override it in elf.h with an undef.  Then you are changing 1 file instead of 4 files.

I'm wondering if any of this is SIFive specific.  The requirement for medany depends on the SiFive memory map, which we inherited from rocket-chip.  Do other RISC-V chip vendors use the same memory map?