riscvarchive/riscv-gcc #169: How can I dump an abstract syntax tree generated by GCC-8.2.0 into a .dot file?

l919898756, 2019-09-19:
The AST data structure has changed a lot in the higher version of GCC. I asked the author of the VCG plugin and the plugin is no longer updated.The AST data structure has changed a lot in the higher version of GCC. I asked the author of the VCG plugin and the plugin is no longer updated.How do I now get the full AST structure printed like the VCG plugin?

@jim-wilson
Hello, long time no see, sorry to bother you again, do you know how I should print the data structure of GCC abstract syntax tree

kito-cheng, 2019-09-19:
`-fdump-tree-all-graph` and `-fdump-rtl-all-graph`, you can got pass dump and dot

l919898756, 2019-09-20:
I want to dump the AST raw internal tree datae, as shown below

l919898756, 2019-09-20:
I would like to know how 
i= i +1 ;
is stored inside the AST. The VCG plug-in can be printed out, but VCG is not applicable to the higher version of GCC

kito-cheng, 2019-09-20:
Hmm, `*.cfg.dot` might be most similar what you want but in gimple, that is generated by `-fdump-tree-all-graph`, so I guess the best way is porting that plug-in into newer version GCC by yourself :P

l919898756, 2019-09-20:
@kito-cheng 
thank you,I'll try it

t-k-, 2021-01-17:
`*.cfg.dot` looks like control flow DAG for SSA, not so relevant to AST. I think clang has a nice way to dump ast `-ast-dump`, but I wish gcc could also support similar thing.