riscvarchive/riscv-gcc #170: GCC generates a call to memcpy to copy into a volatile buffer.

nicber, 2019-10-02:
```
void fill_frame(volatile uint32_t *hw_buffer, frame_header_t header,
                uint32_t *src_buffer, uint16_t length) {
  hw_buffer[0] = ((volatile uint32_t *)&header)[0];
  hw_buffer[1] = ((volatile uint32_t *)&header)[1];
  hw_buffer[2] = ((volatile uint32_t *)&header)[2];
  hw_buffer[3] = (src_buffer[0] << 16 | header.type_length);

  uint32_t *aligned_buffer = (void *)src_buffer + 2;
  for (int i = 0; i < length; ++i) {
    hw_buffer[i + 4] = aligned_buffer[i];
  }
}

int send_frame(frame_header_t header, uint32_t *src_buffer, uint16_t length) {
  if (length > 1500 / 4) {
    length = 1500 / 4;
  }

  update_tx_status();

  if (ping_empty) {
    fill_frame(tx_ping_buff, header, src_buffer, length);
    *tx_ping_length = length + 14;
    tx_ping_control->status = 1;
    ping_empty = 0;
    return length;
  } else if (pong_empty) {
    fill_frame(tx_pong_buff, header, src_buffer, length);
    *tx_pong_length = length + 14 + 2;
    tx_pong_control->status = 1;
    pong_empty = 0;
    return length;
  } else {
    return -1;
  }
}
```
Generates this code:
```
int send_frame(frame_header_t header, uint32_t *src_buffer, uint16_t length) {
800001ac:	fe010113          	addi	x2,x2,-32
800001b0:	00112e23          	sw	x1,28(x2)
800001b4:	00812c23          	sw	x8,24(x2)
800001b8:	00912a23          	sw	x9,20(x2)
800001bc:	01212823          	sw	x18,16(x2)
800001c0:	00050913          	addi	x18,x10,0
800001c4:	00058493          	addi	x9,x11,0
800001c8:	00060413          	addi	x8,x12,0
  if (length > 1500 / 4) {
800001cc:	17700793          	addi	x15,x0,375
800001d0:	00c7f463          	bgeu	x15,x12,800001d8 <send_frame+0x2c>
    length = 1500 / 4;
800001d4:	17700413          	addi	x8,x0,375
  }

  update_tx_status();
800001d8:	ec9ff0ef          	jal	x1,800000a0 <update_tx_status>

  if (ping_empty) {
800001dc:	c281a783          	lw	x15,-984(x3) # 80001828 <ping_empty>
800001e0:	06079663          	bne	x15,x0,8000024c <send_frame+0xa0>
    fill_frame(tx_ping_buff, header, src_buffer, length);
    *tx_ping_length = length + 14;
    tx_ping_control->status = 1;
    ping_empty = 0;
    return length;
  } else if (pong_empty) {
800001e4:	c241a783          	lw	x15,-988(x3) # 80001824 <_edata>
800001e8:	0a078863          	beq	x15,x0,80000298 <send_frame+0xec>
    fill_frame(tx_pong_buff, header, src_buffer, length);
800001ec:	00e00613          	addi	x12,x0,14
800001f0:	00090593          	addi	x11,x18,0
800001f4:	00010513          	addi	x10,x2,0
800001f8:	ed5ff0ef          	jal	x1,800000cc <memcpy>
800001fc:	00040693          	addi	x13,x8,0
80000200:	00048613          	addi	x12,x9,0
80000204:	00010593          	addi	x11,x2,0
80000208:	c101a503          	lw	x10,-1008(x3) # 80001810 <tx_pong_buff>
8000020c:	ec9ff0ef          	jal	x1,800000d4 <fill_frame>
    *tx_pong_length = length + 14 + 2;
80000210:	00040513          	addi	x10,x8,0
80000214:	c0c1a783          	lw	x15,-1012(x3) # 8000180c <tx_pong_length>
80000218:	01040413          	addi	x8,x8,16
8000021c:	0087a023          	sw	x8,0(x15)
    tx_pong_control->status = 1;
80000220:	c081a703          	lw	x14,-1016(x3) # 80001808 <tx_pong_control>
80000224:	00074783          	lbu	x15,0(x14)
80000228:	0017e793          	ori	x15,x15,1
8000022c:	00f70023          	sb	x15,0(x14)
    pong_empty = 0;
80000230:	c201a223          	sw	x0,-988(x3) # 80001824 <_edata>
    return length;
  } else {
    return -1;
  }
}
80000234:	01c12083          	lw	x1,28(x2)
80000238:	01812403          	lw	x8,24(x2)
8000023c:	01412483          	lw	x9,20(x2)
80000240:	01012903          	lw	x18,16(x2)
80000244:	02010113          	addi	x2,x2,32
80000248:	00008067          	jalr	x0,0(x1)
    fill_frame(tx_ping_buff, header, src_buffer, length);
8000024c:	00e00613          	addi	x12,x0,14
80000250:	00090593          	addi	x11,x18,0
80000254:	00010513          	addi	x10,x2,0
80000258:	e75ff0ef          	jal	x1,800000cc <memcpy>
8000025c:	00040693          	addi	x13,x8,0
80000260:	00048613          	addi	x12,x9,0
80000264:	00010593          	addi	x11,x2,0
80000268:	c201a503          	lw	x10,-992(x3) # 80001820 <tx_ping_buff>
8000026c:	e69ff0ef          	jal	x1,800000d4 <fill_frame>
    *tx_ping_length = length + 14;
80000270:	00040513          	addi	x10,x8,0
80000274:	c1c1a783          	lw	x15,-996(x3) # 8000181c <tx_ping_length>
80000278:	00e40413          	addi	x8,x8,14
8000027c:	0087a023          	sw	x8,0(x15)
    tx_ping_control->status = 1;
80000280:	c141a703          	lw	x14,-1004(x3) # 80001814 <tx_ping_control>
80000284:	00074783          	lbu	x15,0(x14)
80000288:	0017e793          	ori	x15,x15,1
8000028c:	00f70023          	sb	x15,0(x14)
    ping_empty = 0;
80000290:	c201a423          	sw	x0,-984(x3) # 80001828 <ping_empty>
    return length;
80000294:	fa1ff06f          	jal	x0,80000234 <send_frame+0x88>
    return -1;
80000298:	fff00513          	addi	x10,x0,-1
8000029c:	f99ff06f          	jal	x0,80000234 <send_frame+0x88>
```

It is my understanding that volatile tells the compiler that it needs to preserve the order of memory accesses and the call to memcpy just discards the qualifier.
The options used for compiling are: `-nostdlib -Og -fstrict-volatile-bitfields -fno-builtin -ffreestanding`
GCC version: 
```
❯ $RISCV/rv32im/bin/riscv32-unknown-elf-gcc --version
riscv32-unknown-elf-gcc (GCC) 9.2.0in/riscv32-unknown-elf-gcc --version — $RISCV/rv32im/bin/riscv32-unknown-elf-gcc```

jim-wilson, 2019-10-07:
The testcase doesn't compile because it is missing types and variables.  But that doesn't seem necessary to see the problem.

It appears that frame_header_t is a structure type.  You are passing this by value to a function, so the code needs to make a copy of it.  It is calling memcpy to make this copy.  Note that the memcpy call is before the fill_frame call, and it is making a copy of header for the call.  A structure copy is an implicit memcpy call.  Perhaps you meant to pass this by address to avoid the copy?

The volatile operations in fill_frame don't appear to have anything to do with this.

nicber, 2019-10-07:
Oh, you are right, I assumed the call to memcpy was due to the volatile calls, since commenting out the body of the function makes gcc not generate a call to memcpy. That must be something else going on.

closed 2019-10-07 by nicber