riscvarchive/riscv-gcc #171: how to install riscv64-unknown-elf-gcc

amitgargmentor, 2019-10-03:
Hi, I downloaded the source code from repository but not able to iinstal. 

i am looking to install gcc for riscv64-unknown-elf-gcc. 

what are the steps ? 
can someone please help ? 

Thanks

kito-cheng, 2019-10-04:
You can use [riscv-gnu-toolchain](https://github.com/riscv/riscv-gnu-toolchain) if you want build from source, or you can download pre-built version from [SiFive](https://www.sifive.com/boards).

Miskatop, 2021-08-12:
You can install riscv64-unknown-elf-gcc in ubuntu using apt : `sudo apt install gcc-riscv64-unknown-elf`

IvanVnucec, 2021-08-20:
@Miskatop Only on Ubuntu 20.04 and up if im not wrong: https://packages.ubuntu.com/search?keywords=gcc-riscv64-unknown-elf

ZhangPeterGree, 2021-10-10:
[image]
Hi guys, I has issue.
compiling error, Can't find heard file. I tried #include <stdio.h> and "stdio.h". but is fatal error.
So can give idea.

Miskatop, 2021-10-11:
@jim-wilson thanks, I successfully build

IvanVnucec, 2021-10-11:
Can you explain what the issue was and how did you managed to fix it?