riscvarchive/riscv-gcc #172: Linker doesn't replace weak functions in assembly with global function in C when LTO is turned on

r901042004, 2019-10-18:
As title, if we have a weak function in _weak.S_.
```asm
.weak func
.type func, @function

.text
func:
    ret
.size func, .-func
```
The corresponding global function is in _strong.c_.
```c
#include <stdio.h>

void func() {
    printf("Hello World!\n");
}
```
Main will call the function in _main.c_.
```c
extern void func();

int main() {
    func();
}
```
If we compile without _-flto_ option, linker will replace the weak function with global function.
The sequence of linking doesn't matter.

```bash
riscv64-unknown-elf-gcc -c weak.S -o weak.o
riscv64-unknown-elf-gcc -c strong.c -o strong.o
riscv64-unknown-elf-gcc -c main.c -o main.o

riscv64-unknown-elf-gcc strong.o weak.o main.o -o main
riscv64-unknown-elf-readelf -a main | grep "func"
# 148: 000000000001015c    28 FUNC    GLOBAL DEFAULT    1 func

riscv64-unknown-elf-gcc weak.o strong.o main.o -o main
riscv64-unknown-elf-readelf -a main | grep "func"
# 148: 000000000001015e    28 FUNC    GLOBAL DEFAULT    1 func
```
If we compile with _-flto_ option, linker will use the function in the later file.

```bash
riscv64-unknown-elf-gcc -flto -c weak.S -o weak.o
riscv64-unknown-elf-gcc -flto -c strong.c -o strong.o
riscv64-unknown-elf-gcc -flto -c main.c -o main.o

riscv64-unknown-elf-gcc -flto strong.o weak.o main.o -o main
riscv64-unknown-elf-readelf -a main | grep "func"
# 54: 000000000001016e     2 FUNC    WEAK   DEFAULT    1 func

riscv64-unknown-elf-gcc -flto weak.o strong.o main.o -o main
riscv64-unknown-elf-readelf -a main | grep "func"
# 23: 000000000001015c    26 FUNC    LOCAL  DEFAULT    1 func
```
* Compiler Version: `GCC 9.2.0`
* Commit: `ed3f6ec269b453f72603e61f617d6d3efffdd8ff` (head of riscv-gcc-9.2.0)

kito-cheng, 2019-10-18:
Hmm, I think it's a more general problem rather than RISC-V specific, I would suggest you drop a mail to gcc-help mailing-list (gcc-help@gcc.gnu.org).

r901042004, 2019-10-20:
Thanks, I've sent email to gcc-help. I also test GCC 8.3.0 which works perfectly.

stefanct, 2020-09-28:
The problem was confirmed upstream: https://gcc.gnu.org/legacy-ml/gcc-help/2019-10/msg00092.html No idea how it continued after that though. Does anybody know if this has been directly addressed upstream and if so which patch fixed this exactly?

jim-wilson, 2020-09-28:
As far as I know, no one ever worked on it.