riscvarchive/riscv-gcc #174: When to support the RISC-V Vector Extension in GCC?

Huracan, 2020-01-05:
Hi, I find Spike has already support Vector Extension, 

i am looking forward risc-v gcc to support Vector Extention as it will help my current work a lot!

Do you have any plan to update recently?

Thanks

kito-cheng, 2020-01-06:
We/SiFive are working on this, and it will release recently :)

Huracan, 2020-01-07:
> 
> 
> We are working on this, and it will release recently :)

ok,thanks a lot

higuoxing, 2020-12-11:
Hi, does GCC support the vector extension now?

kito-cheng, 2020-12-11:
> It looks the vector extension support is in branch riscv-gcc-10.1-rvv-dev?
yes, that the right branch.

it's integrated riscv-gnu-toolchain for that:
https://github.com/riscv/riscv-gnu-toolchain/tree/rvv-intrinsic

higuoxing, 2020-12-11:
> > It looks the vector extension support is in branch riscv-gcc-10.1-rvv-dev?
> 
> yes, that the right branch.
> 
> it's integrated riscv-gnu-toolchain for that:
> https://github.com/riscv/riscv-gnu-toolchain/tree/rvv-intrinsic

Thanks a lot :-)