riscvarchive/riscv-gcc #175: Could not use R_RISCV_PCREL_HI20 to access the .data..percpu section of Linux kernel module

VincentZWC, 2020-01-10:
Romain Dolbeau reported that some modules don't load on RISC-V to the Linux kernel mailing list. The URL of the reported mail is listed below.
https://lore.kernel.org/linux-riscv/1572281840733.3517@european-processor-initiative.eu/#t
This issue is related to the handling of the ".data..percpu" section in the kernel. Hence, please let me quickly describe how the kernel addresses the ".data..percpu" before discussing this issue.

Normally, the kernel continuously places all the module sections in the VMALLOC memory area. However, the ".data..percpu" section is an exceptional case. For convenience, kernel places all the ".data..percpu" sections of the loaded modules in a pre-allocated memory region at runtime. This pre-allocated memory is far, >2G, from the VMALLOC memory area. Hence, it could not be accessed by R_RISCV_PCREL_HI20. 

Let's go back to this problem. Romain declares a static percpu data, test_data, in his kernel module. From the assembly code, I found that it uses R_RISCV_PCREL_HI20 instead of R_RISCV_GOT_HI20 to access the percpu data. I guess that the compiler may think the ".data..percpu" is close to the ".txt" section due to the static attribute.  As mentioned above, the address of ".data.percpu" at runtime is far from the ".txt" section, which is beyond the accessible range of instruction "auipc". Hence, the kernel cannot finish the relocation and reports errors. The assembly code is listed below.
* test.s
  .section        .data..percpu,"aw"
  .align  3
  .set    .LANCHOR0,. + 0
  .type   test_data, @object
  .size   test_data, 16

* objdump -d module.ko
  0000000000000064 <.L0 >:
  64:   00000917                auipc   s2,0x0
  68:   00090913                mv      s2,s2

* readelf -a module.ko
  Relocation section '.rela.init.text' at offset 0xcc600 contains 21 entries:
  Offset          Info           Type           Sym. Value    Sym. Name + Addend
  000000000064  087f00000017 R_RISCV_PCREL_HI2 0000000000000000 .LANCHOR0 + 0

If I remove the "static" attribute from the percpu data declaration, the GCC compiler will use the GOT table to access the percpu data. Hence, the kernel can successfully insert this module.

* test.c
  DEFINE_PER_CPU(uint64_t[2], test_data); 

* objdump -d module.ko
  0000000000000064 <.L0 >:
  64:   00000917                auipc   s2,0x0
  68:   00093903                ld      s2,0(s2) # 64 <.L0 >

*readelf -a module.ko
  Relocation section '.rela.init.text' at offset 0xcc600 contains 21 entries:
  Offset          Info           Type           Sym. Value    Sym. Name + Addend
  000000000064  206000000014 R_RISCV_GOT_HI20  0000000000000000 test_data + 0

I have attached the test.i file, and this case can be reproduced by the following compile instruction.
[test.txt](https://github.com/riscv/riscv-gcc/files/4043914/test.txt)
riscv64-linux-gcc test.i -fPIC -O2 -c -o test_2.o --save-temps

Thanks

aswaterman, 2020-01-10:
The existing PIC code model doesn’t support static data being more than 2GB away from the text section that references it. So this is not unexpected. Perhaps we need to define another code model for kernel modules that forces static variable references to use the GOT...?

Many architectures (other than x86) don’t support 64-bit offsets for PC-relative addressing. How do they handle this case?

VincentZWC, 2020-01-13:
As far as I know, they place the memory region for the kernel module image close to the kernel linear mapping region. Hence, the distance between the pre-allocated .data..percpu region and the module .txt section will not exceed the limit of PC-relative addressing.
( For arm64, the beginning of module head is at 0xffff800008000000 and the pre-allocated .data..percpu region is at 0xffff800011d1a000. )

rdolbeau, 2021-03-11:
Just noticed this issue; a working solution for my particular issue was to move from static to dynamic per-cpu data, which have no issue on RISC-V. That's how I got ZFS working in RV64GC: https://github.com/openzfs/zfs/pull/9942

So this might be a non-issue *if* 'move the module to dynamic per-cpu data' is an acceptable answer.