riscvarchive/riscv-gcc #176: ASM compilation with GCC in GNU Embedded Toolchain — v2019.08.0

ToTamir, 2020-02-21:
Instruction C.LUI should load non-zero 6-bit immediate into bits 17:12, clear the bottom 12 bits and sign-extends bit 17 into all higher bits.
But:
C.LUI x1, 0x3f doesn't work (Error: illegal operands),
C.LUI x1, -0x20 doesn't work (Error: illegal operands),
C.LUI x1, 0x20 doesn't work (Error: illegal operands),
C.LUI x1, -0x1 doesn't work (Error: illegal operands).
The highest possible value accepted by compiler is 0x1f (5-bit immediate with all bits set). The lowest possible value accepted by compiler is 0x01 (0x0 is reserved and negative values are not accepted).
So C.LUI in your compiler loads only 5-bit immediate and not sign-extends it.

Instruction C.ADDI4SPN should add zero-extended non-zero immediate, scaled by 4, to stack pointer (x2), and write result to rd'. That's mean this instruction have only two parameters: immediate and rd'.
But:
C.ADDI4SPN x8, 0x4 doesn't work (Error: illegal operands).
Compiler require C.ADDI4SPN x8, x2, 0x4. Binary file after compilation is good.

jim-wilson, 2020-02-21:
The main problem here is that we lack good assembler documentation.  Which means people either have to guess at the assembly syntax or else read the binutils sources to figure it out.  The riscv/riscv-asm-manual repo is supposed to be an assembler manual but no one has ever filled it out.

The intent here is that people write non-C instructions,  and the assembler automatically converts non-C instructions to C instructions when rvc is enabled when possible.  So it isn't necessary to use C instructions if you are hand writing assembly code.  This is only necessary if you want to test hardware or assembler support for C instructions.

c.lui uses the same syntax as lui for consistency.  So it only accepts 20-bit unsigned numbers in range of the signed-6 bit immediate.
        .option rvc
        c.lui a0, 0x0001f
        c.lui a0, 0x00001
        c.lui a0, 0xfffff
        c.lui a0, 0xfffe0

c.addi4spn accepts the same arguments as addi, it has a source (sp) and a dest, so it takes three arguments, with sp as one of the arguments.  c.addi16sp has a source (sp) and dest (sp) that are the same, so it takes two arguments, with sp as one of the arguments.

There isn't really anything wrong with assembler syntax here.  It just isn't documented well enough.

ToTamir, 2020-02-24:
I used C.LUI x10, 0xfffff instead of C.LUI x10, 0x3f or C.LUI x10, -0x1 and this works now properly. Your explanation of C.ADDI4SPN syntax sounds logical for me. So thank you very much.

closed 2020-02-24 by ToTamir