riscvarchive/riscv-gcc #178: RISC-V GCC does not follow GCC convention for __builtin___clear_cache()

lbmeng, 2020-03-11:
The GCC document [1] has the following description about __builtin___clear_cache():

> Built-in Function: void __builtin___clear_cache (void *begin, void *end)
> This function is used to flush the processor’s instruction cache for the region of memory between begin inclusive and end exclusive. Some targets require that the instruction cache be flushed, after modifying memory containing code, in order to obtain deterministic behavior.
> 
> If the target does not require instruction cache flushes, __builtin___clear_cache has no effect. Otherwise either instructions are emitted in-line to clear the instruction cache or **a call to the __clear_cache function in libgcc is made**.

It says a call to __clear_cache() function should be generated to replace __builtin___clear_cache().
This was how it was done on GCC ARM and ARM64:

GCC ARM: (in gcc/config/arm/arm.c)
```
  emit_library_call (gen_rtx_SYMBOL_REF (Pmode, "__clear_cache"),
                     LCT_NORMAL, VOIDmode, a_tramp, Pmode,
                     plus_constant (Pmode, a_tramp, TRAMPOLINE_SIZE), Pmode);
```

GCC ARM64: (in gcc/config/aarch64/aarch64.c)
```
  emit_library_call (gen_rtx_SYMBOL_REF (Pmode, "__clear_cache"),
                     LCT_NORMAL, VOIDmode, a_tramp, ptr_mode,
                     plus_constant (ptr_mode, a_tramp, TRAMPOLINE_SIZE),
                     ptr_mode);

```

But GCC RISC-V (in gcc/config/riscv/riscv.md) generates call to __riscv_flush_icache(), which does NOT follow GCC's documentation about __builtin___clear_cache().
```
  emit_library_call (gen_rtx_SYMBOL_REF (Pmode, ICACHE_FLUSH_FUNC),
                     LCT_NORMAL, VOIDmode, operands[0], Pmode,
                     operands[1], Pmode, const0_rtx, Pmode);
```
ICACHE_FLUSH_FUNC is defined in gcc/config/riscv/linux.h:
`#define ICACHE_FLUSH_FUNC "__riscv_flush_icache"`

This to me GCC RISC-V port did not follow the convention as used on other architectures.

[1] https://gcc.gnu.org/onlinedocs/gcc/Other-Builtins.html

aswaterman, 2020-03-11:
The RISC-V port is doing the right thing for RISC-V. This is ultimately an
implementation detail (after all, x86 doesn’t even have a nontrivial
implementation of this routine). It is not a problem that our
implementation differs from other architectures in this respect, because
other popular architectures already differ from each other.

Given the context that I assume led you to raise this issue, my response
is: fix the software you’re working on, rather than trying to push the
issues into GCC and the RISC-V ABI.

lbmeng, 2020-03-11:
> The RISC-V port is doing the right thing for RISC-V. This is ultimately an implementation detail (after all, x86 doesn’t even have a nontrivial implementation of this routine). It is not a problem [...]

Yes, this issue can be simply fixed or workarounded in the software by providing __riscv_flush_icache(). However the behavior of __builtin___clear_cache() in the GCC documents as I pointed out is NOT followed by RISC-V GCC port.

If you think RISC-V GCC is doing the right thing, then fix the GCC documents.

And I do not see this is mentioned in the RISC-V ABI doc. Maybe you were referring to another ABI doc I am not aware of?
https://github.com/riscv/riscv-elf-psabi-doc/blob/master/riscv-elf.md

aswaterman, 2020-03-11:
I don’t mean to be flippant, but I just googled for __riscv_flush_icache
and the first hit was quite relevant:
https://www.gnu.org/software/libc/manual/html_node/RISC_002dV.html

lbmeng, 2020-03-11:
> I don’t mean to be flippant, but I just googled for __riscv_flush_icache and the first hit was quite relevant: https://www.gnu.org/software/libc/manual/html_node/RISC_002dV.html

I saw this document before, but that does not answer my question. In my original post, I was referring to GCC document @ https://gcc.gnu.org/onlinedocs/gcc/Other-Builtins.html.

My question was to whether GCC RISC-V port **correctly** implements __builtin___clear_cache(), not about some other random built-in functions like __riscv_flush_icache().

```
E.2 RISC-V-specific Facilities
Cache management facilities specific to RISC-V systems that implement the Linux ABI
are declared in sys/cachectl.h.

Function: void __riscv_flush_icache (void *start, void *end, unsigned long int flags)
```

From the document above, it does not mention this function's connection to __builtin___clear_cache().

aswaterman, 2020-03-11:
I see your point: the generic GCC docs are technically incorrect here. It’s
a very minor point, though.  One of the options the doc mentions is
“instructions
are emitted in-line to clear the instruction cache”. There is a very fine
line between “jal __riscv_flush_icache” and “instructions are emitted
in-line”—specifically; if we delete the word “in-line”, the docs would
become correct.

Feel free to file a bug at FSF requesting a documentation improvement or
send in a patch that improves the docs.

lbmeng, 2020-03-11:
> I see your point: the generic GCC docs are technically incorrect here. It’s a very minor point, though. One of the options the doc mentions is “instructions are emitted in-line to clear the [...]

So currently GCC RISC-V always translate call to __builtin___clear_cache() into __riscv_flush_icache() with flags being zero which means a global icache flush.

I have not checked with LLVM yet, but if this is determined by whatever RISC-V compiler convention (to me this is not ABI) on treating __builtin___clear_cache(), I believe LLVM has to do the same to keep compatibility with GCC.

> Feel free to file a bug at FSF requesting a documentation improvement or send in a patch that improves the docs.

Okay, will do.

aswaterman, 2020-03-11:
For better or worse, it is very much an ABI thing: libc.so provides __riscv_flush_icache, and so the compiler’s convention is not merely internal. Therefore, clang must do the same thing if it intends to work with glibc.

lbmeng, 2020-03-11:
Yep we can say it's a Linux ABI because __riscv_flush_icache() is provided in glibc. Not a RISC-V ABI though.

lbmeng, 2020-03-12:
A bug was filed to GCC bugzilla
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=94136

Close this issue.

closed 2020-03-12 by lbmeng