riscvarchive/riscv-gcc #181: Question about the stack alignment when subprocedure calling exists

Engineev, 2020-04-13:
Hello. When I try to use the command `riscv32-unknown-linux-gnu-gcc -S  -mabi=ilp32` to compile the following C code
```
int main() {
  int a;
  scanf("%d", &a);
  return a;
}
```
it produces assembly code like this one
```
main:
	addi	sp,sp,-32
	sw	ra,28(sp)
	addi	a1,sp,12
	lui	a0,%hi(.LC0)
	addi	a0,a0,%lo(.LC0)
	call	__isoc99_scanf
	lw	a0,12(sp)
	lw	ra,28(sp)
	addi	sp,sp,32
	jr	ra
```

I wonder why the first line in `main` is `addi sp, sp, -32` instead of something like `addi sp, sp, -16` since it seems that the specification only requires a 16-byte alignment. Am I missing something here, or it's just some implementation choice.

aswaterman, 2020-04-13:
Only 16-byte alignment is required.  In principle, it would be OK to allocate `ra`'s save slot and the `a` variable next to each other, reducing the frame size to 16 bytes.

It looks like our GCC implementation is conservatively forcing 16-byte alignment between the local-variable area and the register-save area.  It's probably possible to reduce this alignment in some cases, since the register-save area only needs to be aligned to the size of the largest register saved.  But there are probably some gotchas there, and varargs functions might be one of the gotchas, so it might not help in the `scanf` case anyway.

Engineev, 2020-04-13:
Thank you for your explanation!

closed 2020-04-13 by Engineev