riscvarchive/riscv-gcc #182: Support for ZFinx extension

longhanlie, 2020-04-23:
In order to support Zfinx extension ( X register replaces F register to store data), In the initial implementation plan, several issues need to be discussed:
1. What form of the target need to be added, -march = imfzc or -march = imfc_zfinx ?
2. When objdump is disassembled, since the opcode of the original f instruction is not changed, and the content of march is not recognized, the F register cannot be directly disassembled into the X register.
At present, -Memit-zfinx is added as an option to achieve disassembly printing of the X register logo, but it isn't elegant. Is there another way to use the flag to disassemble the X register ?
3. Can we add relevant march information to the ELF file for reading, as a disassembly flag for printing the X register ? Or adding a disassembler options is enough, just like the current implementation.

kito-cheng, 2020-04-23:
> 1. What form of the target need to be added, -march=imfzc or -march=imfc_zfinx ?
Latter one, it should be -march=rv*imfc_zfinx

> 2. When objdump is disassembled, since the opcode of the original f instruction is not changed, and the content of march is not recognized, the F register cannot be directly disassembled into the X register.
At present, -Memit-zfinx is added as an option to achieve disassembly printing of the X register logo, but it isn't elegant. Is there another way to use the flag to disassemble the X register ?

There is ELF attribute[1] to record march, but objdump didn't reference that yet, we are implement that recently, however ELF attribute not enable by default, so I think it worth to add new option -March= to objdump.

> 3. Can we add relevant march information to the ELF file for reading, as a disassembly flag for printing the X register ? Or adding a disassembler options is enough, just like the current implementation.

That's what ELF attribute do,  LLVM and GNU toolchain are both supported now, you can observe that by `riscv64-unknown-elf-readelf -A`

e.g:
```
$ riscv64-unknown-elf-readelf -A a.out 
Attribute Section: riscv
File Attributes
  Tag_RISCV_stack_align: 16-bytes
  Tag_RISCV_arch: "rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0"
```

[1] https://github.com/riscv/riscv-elf-psabi-doc/pull/71

longhanlie, 2020-04-24:
Whether to add a new option (-March =) to objdump is in schedule ?

longhanlie, 2020-04-24:
@kito-cheng Whether to add a new option (-March =) to objdump is in schedule ?

kito-cheng, 2020-04-28:
We/SiFive don't have particular schedule on that, so maybe need to wait someone in the community 
 have free time to do that.