riscvarchive/riscv-gcc #183: picojpeg fails with -march=rv32imacb but passes with rv32imac

eflaner, 2020-04-27:
### Issue: 
**picojpeg** fails to run when compiled with _-march=**rv32imacb**_ whereas the same succeeds in _-march=**rv32imac**_

### Steps to reproduce error:

1. Unzip the files [picojpeg.zip](https://github.com/riscv/riscv-gcc/files/4540258/picojpeg.zip)
2. run 
`riscv32-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -mcmodel=medany  -fdata-sections -ffunction-sections -Os -Wl,-gc-sections -nostartfiles -DCPU_MHZ=1 -T./link.ld -o picojpeg -I./ crt.S syscalls.c picojpeg.c picojpeg_test.c`
3. run
`spike --isa=rv32imac picojpeg`
4. The output should be 
> returned result =0

Now rerun step 2 by replacing **-march=rv32imacb** and **run** 
`spike --isa=rv32imacb picojpeg `
The output is shown as 
> Trap Cause: 5 encountered on PC:8000175e.

kito-cheng, 2020-04-28:
B extension is not officially support yet, but maybe @cliffordwolf or @mablinov could help you.

mablinov, 2020-05-01:
@eflaner  Can't seem to replicate on my end.

Although I'll note that in both cases, the `returned result  = 0` output gets mangled (the first `r` character is missing and the whole output is shifted forward in the terminal window.) Maybe thats just WSL.

eflaner, 2020-05-01:
@mablinov 

here are the commit id's of the toolchain build.

- riscv-gcc               -  d69e3ef
- riscv-newlib          - 5726873
- riscv-binutils-gdb - c870418
- riscv-isa-sim        -  5b042d7

Could you please let me know if it's same.

jim-wilson, 2020-05-20:
I was able to reproduce.  I tracked it down to a register allocator bug that shows up in the 278.reload rtl dump file.  Since this is an FSF GCC development tree from last Sept, this bug isn't interesting.  It was probably already fixed in the FSF tree months ago.  I can make the testcase work with a few minor tweaks to work around the register allocator bug.
```
diff -u orig/picojpeg.c tmp/picojpeg.c
--- orig/picojpeg.c	2020-04-27 07:08:30.000000000 -0700
+++ tmp/picojpeg.c	2020-05-19 17:25:03.297878251 -0700
@@ -600,7 +600,7 @@
 }
 
 //------------------------------------------------------------------------------
-static uint8
+static uint8 __attribute__ ((noinline))
 readDHTMarker (void)
 {
   uint8 bits[16];
@@ -613,7 +613,8 @@
 
   while (left)
     {
-      uint8 i, tableIndex, index;
+      volatile uint8 index;
+      uint8 i, tableIndex;
       uint8 *pHuffVal;
       HuffTable *pHuffTable;
       uint16 count, totalRead;
```

closed 2020-06-04 by eflaner