riscvarchive/riscv-gcc #184: Assembly code for RISC-V division does not match RISC-V spec in case of division by zero

virginiemoser, 2020-05-26:
The assembly code in libgcc/config/riscv/div.S does not handle the division by zero as specified in the riscv-spec v2.2 chapter 6.2 in case of signed division:

> The quotient of division by zero has all bits set, i.e.  2XLEN−1 for unsigned division or−1 for signed division.

When a negative number is divided by zero in the __divdi3 function, the result is 1 instead of -1.

The following patch should resolve the issue:

`diff --git a/libgcc/config/riscv/div.S b/libgcc/config/riscv/div.S `
`index 922a4338042..57f5856e11d 100644 `
`--- a/libgcc/config/riscv/div.S `
`+++ b/libgcc/config/riscv/div.S `
`@@ -107,7 +107,8 @@ FUNC_END (__umoddi3) `
`   /* Handle negative arguments to __divdi3.  */`
` .L10:`
`   neg   a0, a0`
`-  bgez  a1, .L12      /* Compute __udivdi3(-a0, a1), then negate the result.  */`
`+  bgtz  a1, .L12      /* Compute __udivdi3(-a0, a1), then negate the result.  */`
`+                      /* Zero is handled as a negative so that the result will not be inverted */`
`   neg   a1, a1`
`   j     __udivdi3     /* Compute __udivdi3(-a0, -a1).  */`
` .L11:                 /* Compute __udivdi3(a0, -a1), then negate the result.  */ `

aswaterman, 2020-05-26:
The spec for __divdi3 isn’t that it matches the RISC-V div instruction;
it’s that it matches the C operator /, where this result is unspecified. So
I think it’s OK as-is.

jim-wilson, 2020-05-26:
If you want to submit a patch, it must be sent upstream.  Also, FSF GCC patches require copyright assignments, though we can accept a patch without an assignment if it is small enough and trivial enough.  It is OK to report bugs here.

I agree with Andrew that it doesn't have to exactly match the hardware for edge conditions, but this does mean that -mno-div and -mdiv may give different results which could potentially be a problem.  Though it is also the case that soft-float and hard-float don't give exactly the same results in some cases, and this is generally not a problem.

jim-wilson, 2020-06-02:
This patch was subimitted upstream, and is being handled there.