riscvarchive/riscv-gcc #190: internal compiler error: in riscv_gpr_save_operation_p, at config/riscv/riscv.c:5219

ppannuto, 2020-06-15:
I'm happy to upstream this to gcc proper if needed, but since the brew formula pulls the toolchain from here, I figured this might be the place to start.

The error is basically as described in the title. The same build works fine with an ARM toolchain.

```
$ riscv64-unknown-elf-gcc -std=gnu11 -Wbad-function-cast  -Wjump-misses-init  -Wmissing-prototypes  -Wnested-externs  -Wold-style-definition  -frecord-gcc-switches -gdwarf-2 -Os -fdata-sections -ffunction-sections -fstack-usage -Wstack-usage=2048 -Wall -Wextra -Wl,--warn-common -Wl,--gc-sections -Wl,--emit-relocs -fPIC -include ../../support/warning_header.h -Wdate-time  -Wfloat-equal  -Wformat-nonliteral  -Wformat-security  -Wformat-y2k  -Winit-self  -Wlogical-op  -Wmissing-declarations  -Wmissing-field-initializers  -Wmissing-format-attribute  -Wmissing-noreturn  -Wmultichar  -Wpointer-arith  -Wredundant-decls  -Wshadow  -Wtrampolines  -Wunused-macros  -Wunused-parameter  -Wwrite-strings  -I../../libtock/ -I../../libtock//include -I../../libtock//../include -I../../libtock/internal/ -I../../libtock/internal//include -I../../libtock/internal//../include -march=rv32imac -mabi=ilp32 -mcmodel=medlow -Wl,--no-relax    -c -o ../../libtock/build/rv32imac/tock.o ../../libtock/tock.c -save-temps
during RTL pass: ira
../../libtock/tock.c: In function 'yield':
../../libtock/tock.c:165:1: internal compiler error: in riscv_gpr_save_operation_p, at config/riscv/riscv.c:5219
  165 | }
      | ^
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
```

[tock.i.zip](https://github.com/riscv/riscv-gcc/files/4780293/tock.i.zip) (GitHub won't allow the plain `.i` but will take the zip of it? Sorry)

---

To reproduce, simply run `make` in this folder (`blink`) of this branch (`riscv-for-all`): https://github.com/tock/libtock-c/tree/riscv-for-all/examples/blink.
By default, the build system is polyglot, to just build for riscv and get the verbose output: `TARGETS=rv32imac TOCK_ARCHS=rv32imac make V=1`

kito-cheng, 2020-06-15:
Thanks for report this, I'll fix this soon on upstream and backport to here :)

kito-cheng, 2020-06-15:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=95683

closed 2020-06-19 by kito-cheng