riscvarchive/riscv-gcc #194: loop optimization for static prediction

gxflying, 2020-08-26:
Hi，I find that for loop control the gcc will always emit the branch instruction at the end of the code, which means that the branch will always be taken  while the condition control always emits branch instructions NOT taken 
I think if we modified the loop PASS to emit NOT taken branch first,  it will be better for statical branch prediction
bellow with -O3 option
[image]

gxflying, 2020-08-26:
[image]

aswaterman, 2020-08-26:
It doesn't make sense to employ this strategy by default, because it will increase code size and de-optimize the code for most processors.  This would be a microarchitectural-specific optimization enabled with `-mtune=your_processor_name_here`.

Even for simple processors, it's not a clear win.  Since there's now an additional instruction in the loop, you'd only see a perf improvement if the branch-taken penalty is two or more cycles.  If the pipeline is deep enough for branches to resolve in the 3rd pipeline stage, then the cost of a 1-entry branch-target buffer is very minor relative to the other hardware.  Adding the 1-entry BTB and using the original codegen strategy will perform better, and (for this loop) will almost certainly use less energy per loop iteration.

gxflying, 2020-08-26:
thank for your reply
I agree that we should NOT set this as a default strategy.  and this may cause a little bigger code size.
but think about the senario of the riscv mcu, where always have a tie area budget.  it will always not contains a simple statical predict strategy like "always taken" or "always not taken"
when the loop and the if statement can both emit instructions with the same taken conditions as I addressed above it will have a better IPC but a bigger code size.
i think it maybe better that gcc have a option to let user decide weather to enable this optimization.
or do we already have this feature?

aswaterman, 2020-08-26:
You are always free to build an optimizer for your processor and offer it for upstream support. But I’ll remark again that the design doesn’t really make sense, even in the MCU market, for reasons I explained in my previous post.

gxflying, 2020-08-26:
yeah,  maybe I should do some test on what you said about the perf improvement
for now I will close this issue.
thank you very much ~

closed 2020-08-26 by gxflying