riscvarchive/riscv-gcc #195: Will KEEP command work when --ffunction-sections flag is enabled?

SaiVK, 2020-09-12:
Hello Everyone,
In order to understand **function-sections** I had enabled the _--ffunction-sections_ flag. As expected, each function was placed in its own text section. Also, unreferenced functions were eliminated from the final executable created. 

For those eliminated functions I used the _**KEEP**_ linker command explicitly so that they are present in the final executable. But those functions were still eliminated from the final executable.
Note that I had not used _--gc-sections_ flag in the linker.

So does this mean, **_KEEP_** linker command does not work when _--ffunction-sections_ flag is enabled?
If yes, is there any command that will prevent the Linker from eliminating unreferenced functions even when the _--function-settings_?

I hope I did not misunderstand anything.

Thanking you
Sai

jim-wilson, 2020-09-12:
If you aren't using --gc-sections, then nothing will be eliminated at link time.  Unless perhaps you are also using link time optimization (LTO), in which case it is actually the compiler that is optimizing away functions, and linker script commands like KEEP are irrelevant, because the compiler doesn't look at linker scripts.  If this is the case, then adding __attribute__ ((used)) to functions will prevent them from being optimized away.  There is also another way to explain this using archive (.a) files, but I think LTO is more likely.  If you are using LTO, then the -ffunctions-sections flag is irrelevant.  Functions will be optimized away even without that option with LTO.

I'm guessing here, because you haven't provided a complete example, with enough info to reproduce.  There are a lot of compiler and linker options and features that interact in complex ways.  If you don't mention enough info to reproduce, then I have to guess what you are doing.

KEEP does work with --gc-sections, but you said you aren't using --gc-sections.

SaiVK, 2020-09-12:
Hello Jim
  I am building an archive file (.a) file using the -ffunction-sections flag. The LTO option is not enabled. Only flag that is enabled is -mcmodel=medany. The missing functions are part of the archive file.

jim-wilson, 2020-09-12:
Object files are only pulled from an archive file if there is a reference to a function in the object file.  This is the point of an archive file.  You can put one hundred object files in there, and if you only need one of them to satisfy a link, only one is pulled out into your executable.  If the issue is how archive files work though, then the -ffunction-sections option should not be changing the result.  You should get the same result with and without -ffunction-sections.  There could still be some other unknown issue affecting your result.

If you want to include all of the object files in an archive, then you can use --whole-archive.  But this will make your executables much larger, since you will be pulling in a lot of unused code from the archive file.  You can also use --no-whole-archive to get the default behavior back.  if linking with more than one library, and you only want one library to have special treatment, then surround that library with --whole-archive/--no-whole-archive options.

SaiVK, 2020-09-13:
Hello @jim-wilson 

Apologies, I had a look at the Archive file that is getting generated. All the functions are present in the archive file, built using _-ffunction-sections_.
But when I link the library with other object files to create the final executable using the custom Linker script with _KEEP_ linker command, then unreferenced functions are eliminated from the final executable/object file generated.

This is the final command I used to create the executable based on the library file: _libboot.a_
`riscv64-unknown-elf-gcc -mcmodel=medany hello -o ${output_dir}/${kernel_name} -static -nostdlib -nostartfiles -Wl,-T ${LINKER_PATH}/linker.x -Wl,--start-group libboot.a -Wl,--end-group`

This is my Linker Script: I am using the _KEEP_ linker command explicitly so that those functions are not removed.
`/* taken (and modified) from riscv-pk. See LICENSE.riscv-pk */

OUTPUT_ARCH( "riscv" )

ENTRY( reset_vector )

SECTIONS
{
    . = 0x80000000;

    .text : {
         KEEP(*(.text.init))

          KEEP(*(.text.boot_primary))
          KEEP(*(.text.boot_secondary))
          KEEP(*(.text.timer_pending))
          KEEP(*(.text.set_timer))
          KEEP(*(.text.clear_timer))

		 
         *(.text)
         *(.text.*)
        /* *(.gnu.linkonce.t.*) */
    }

  . = ALIGN(16);
/* rodata: Read-only data */
  .rodata :
  {
    PROVIDE( _srodata = . );
    *(.rdata)
    *(.rodata)
    *(.rodata.*)
    *(.gnu.linkonce.r.*)
    PROVIDE( _erodata = . );
  }

  /*--------------------------------------------------------------------*/
  /* HTIF, isolated onto separate page                                  */
  /*--------------------------------------------------------------------*/
  . = ALIGN(0x1000);
  .htif :
  {
    PROVIDE( __htif_base = . );
    *(.htif)
  }
  . = ALIGN(0x1000);

  /*--------------------------------------------------------------------*/
  /* Initialized data segment                                           */
  /*--------------------------------------------------------------------*/

  /* Start of initialized data segment */
  . = ALIGN(16);
   _fdata = .;

  /* data: Writable data */
  .data :
  {
  *(.data)
  *(.data.*)
  *(.srodata*)
  *(.gnu.linkonce.d.*)
  *(.comment)
  }

  /* End of initialized data segment */
  PROVIDE( edata = . );
  _edata = .;

  /* Have _gp point to middle of sdata/sbss to maximize displacement range */
  . = ALIGN(16);
  _gp = . + 0x800;

  /* Writable small data segment */
  .sdata :
  {
    *(.sdata)
    *(.sdata.*)
    *(.srodata.*)
    *(.gnu.linkonce.s.*)
  }

  /*--------------------------------------------------------------------*/
  /* Uninitialized data segment                                         */
  /*--------------------------------------------------------------------*/

  /* Start of uninitialized data segment */
  . = ALIGN(8);
  _fbss = .;

  /* Writable uninitialized small data segment */
  .sbss :
  {
    *(.sbss)
    *(.sbss.*)
    *(.gnu.linkonce.sb.*)
  }

  /* bss: Uninitialized writeable data section */
  . = .;
  _bss_start = .;
  .bss :
  {
    *(.bss)
    *(.bss.*)
    *(.gnu.linkonce.b.*)
    *(COMMON)
  }
  _bss_end = .;
  _ebss = .;

  _end = ALIGN(8);
  /* End of uninitialized data segment
   * used for heap allocation
   */ 
  PROVIDE( __KERNEL_END = . );
  __KERNEL_END = .;
}
`
Apologies for not providing all the details earlier.

Thank you so much 
Sai

closed 2021-04-04 by SaiVK