riscvarchive/riscv-gcc #196: How to insert a custom instruction at the start of every function by default?

SaiVK, 2020-09-13:
Hello Everyone
   I have created one new instruction that has to be inserted at the start of every function at compile time. I have implemented the instruction in _spike_. I want to modify the riscv-gcc compiler such that the instruction gets inserted by default for all the function.

Should I change the _riscv_expand_prologue()_ in _riscv-gcc/gcc/config/riscv/riscv.c_ to achieve this?
Any help regarding this would help me.

Thank you so much
-Sai

kito-cheng, 2020-09-14:
> Should I change the riscv_expand_prologue() in riscv-gcc/gcc/config/riscv/riscv.c to achieve this?
It's a good start point, you could add your instruction pattern to md file and try to add emit_insn in riscv_expand_prologue.

SaiVK, 2020-09-15:
Thanks, @kito-cheng. I will try doing that.