riscvarchive/riscv-gcc #197: Broken compilation result in O2 mode

zetalog, 2020-09-23:
[image]
[image]

Can clearly see compilation breakage in kernel/firmware builds.
Version 10.1.0, recent upstream (several days ago).

aswaterman, 2020-09-23:
GCC is replacing a code sequence that's equivalent to `memset` with a call to `memset`.  Unfortunately, it's doing so within a `memset` routine, so the effect is an infinite loop.

You can suppress this behavior with the `-fno-builtin` flag.

Note this isn't a RISC-V-specific issue; I see the same behavior for x86-64 GCC.

jim-wilson, 2020-09-23:
This can also be fixed with -fno-tree-loop-distribute-patterns which has a more limited effect than -fno-builtin, and is what newlib uses for all targets.

zetalog, 2022-06-22:
Closing with recent version of riscv-gnu-toolchains.

closed 2022-06-22 by zetalog