riscvarchive/riscv-gcc #198: question about the behavior to generate c.jal or jal ?

gxflying, 2020-09-28:
Hi sir, I am doing some verifications on riscv  "C" extension,  but some thing about the c.jal makes me confused.
when I set the jump offset to 0x7f6 which is a legal value for c.jal,  the gcc give me a  jal instruction?  
please Note that I explicitly used c.jal in my disassembly code and compiling with option -march=rv32imac
I think this should be a "c.jal" after being compiled
and is there some rules to cliam that when to generate a "c.jal" or a "jal" ?
thanks~
[image]
[image]

gxflying, 2020-09-28:
and according to the c.jal coding format the c.jal pc+0x7f6 should be "0x2FDD" !

[image]

jim-wilson, 2020-09-28:
I can't read your example.  Text is better than pictures.

The linker needs to allow some wiggle room due to alignment and other issues that can cause branches to be accidentally out of range, so the linker typically is not able to use the full range of branches.

gxflying, 2020-09-28:
thanks for your reply and sorry the tehe picture:
code sample below the address in comment is from disassembly code:
**main:**
.align 12              # align to 0x80009000
nop                     #advance PC to 0x80009004
nop                     #advance PC to 0x80009008
c.addi x0, 0x01    #advance PC to 0x8000900a
**c.jal offset_7fe     #jump to 0x80009808**
.align 11              # align to 0x80009800
nop                     #advance PC to 0x80009804
nop                     #advance PC to 0x80009808
**offset_7fe:**           # label at 0x80009808
do something

the **c.jal offset_7fe** turns out to be a "jal"  after being compiled  which is 0x7fe000ef  
and it can be a compressed instruction : 0x2FFD
[image]

jim-wilson, 2020-09-28:
The toolchain doesn't support what you are trying to do.  Relxaing branches is complicated, and we can't always give ideal reaults.