riscvarchive/riscv-gcc #202: Escaping operators as part of section names in Linker script

SaiVK, 2020-10-24:
Hello Everyone
    I am generating custom linker scripts and optimizing the placement of code sections. Few of the code section names that I generate have operators as part of their names: like '_>_', '_<_', '_&_' etc...
eg: *(.text.code_1&234)
I am tried escaping the operators using _\_ but it was not working. Any other suggestions will help me a lot.

Thank you so much
Sai

aswaterman, 2020-10-24:
This issue tracker is meant for issues specific to the RISC-V GCC port.  This isn't a RISC-V issue or a GCC issue.

I'm not a linker script expert, but I'm unaware of a mechanism to escape special characters.  You might get away with replacing them with something non-special, e.g., replace all non-alphanumeric characters (including `_`) with `_X_`, where X is the numeric value of the byte.  But you're probably better off avoiding special characters altogether; it seems like you're creating extra work for yourself.

SaiVK, 2020-10-24:
Apologies for raising this issue over here. Please excuse me.
Thank you so much for your suggestion. 

-Sai

closed 2020-10-25 by SaiVK