riscvarchive/riscv-gcc #207: Undefined reference to `__udivdi3' when enabling LTO

DurandA, 2020-11-01:
When [enabling LTO](https://github.com/DurandA/litex/commit/15a6fc038727b3782870bdba04f5cae712afe79c) to compile the [LiteX](https://github.com/enjoy-digital/litex) bios, it GCC triggers an undifned reference to `__udivdi3`:

```
riscv64-unknown-elf-gcc -std=gnu99 -nostdlib -nodefaultlibs -Os -mno-save-restore -march=rv32im     -mabi=ilp32 -D__picorv32__  -L/home/duranda/devel/build/software/include -T /home/duranda/devel/litex/litex/soc/software/bios/linker.ld -N -o bios.elf \
	../libbase/crt0.o \
	isr.o boot-helper.o boot.o helpers.o cmd_bios.o cmd_mem.o cmd_boot.o cmd_i2c.o cmd_spiflash.o cmd_litedram.o cmd_liteeth.o cmd_litesdcard.o main.o complete.o readline.o \
	-L../libcompiler_rt \
	-L../libbase \
	-L../liblitedram \
	-L../libliteeth \
	-L../liblitespi \
	-L../liblitesdcard \
	-llitedram -lliteeth -llitespi -llitesdcard -lbase-nofloat -lcompiler_rt
/home/duranda/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/bios.elf.OQpKqZ.ltrans0.ltrans.o: in function `.L0 ':
/home/duranda/devel/litex/litex/soc/software/bios/cmds/cmd_mem.c:197: undefined reference to `__udivdi3'
/home/duranda/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/bios.elf.OQpKqZ.ltrans0.ltrans.o:/home/duranda/devel/litex/litex/soc/software/bios/cmds/cmd_mem.c:209: undefined reference to `__udivdi3'
collect2: error: ld returned 1 exit status
make: *** [/home/duranda/devel/litex/litex/soc/software/bios/Makefile:64: bios.elf] Error 1
```

From what I understand, `__udivdi3` is [part of libgcc and is used as a fallback when types are not supported natively](https://stackoverflow.com/a/34533569/1584915). I also saw that the flag `-nodefaultlibs` excludes libgcc. However, the compilation succeed with the same set of flags when LTO is not enabled.

Is this an expected behaviour?

aswaterman, 2020-11-01:
If LTO makes a different optimization decision than no-LTO, it could affect which libgcc routines are needed.  This one's a little surprising, but I can imagine situations that could make it happen.  I recommend linking against libgcc explicitly.

DurandA, 2020-11-01:
@aswaterman Linking against gcc explicitly solves the issue.

> If LTO makes a different optimization decision than no-LTO, it could affect which libgcc routines are needed.

Thanks, this explanation is very useful.

closed 2020-11-01 by DurandA