riscvarchive/riscv-gcc #213: Soft-float code generation

georgeto, 2020-11-10:
Is it possible to prevent the RISC-V gcc compiler from using floating-point registers/instructions without imposing the same restrictions on the assembler? (see [-mgeneral-regs-only](https://gcc.gnu.org/onlinedocs/gcc/AArch64-Options.html#mgeneral-regs-only) for ARM64 or [-msoft-float](https://gcc.gnu.org/onlinedocs/gcc/MIPS-Options.html#MIPS-Options) for MIPS)

On RISC-V we can control floating-point code generation via the `-mabi` and `-march` flags, but there seems to be no combination that leads to the desired behavior.
`-mabi` only controls the floating-point calling convention, but still allows the compiler to generate floating-point instructions.
`-march` without the F or D extensions prevents the compiler from generating floating-point instructions, but imposes the same restriction on the assembler.

aswaterman, 2020-11-10:
You might be able to do something like `-mabi=lp64 -march=rv64imac -Wl,-march=rv64imafdc`, but I haven't tested it.

georgeto, 2020-11-10:
Seems to work, thank you!

closed 2020-11-10 by georgeto