riscvarchive/riscv-gcc #220: riscv gcc does not support common block mechanism?

unicornx, 2020-12-03:
tested with following source code:
```
$ cat 1.c
char c;
void main()
{
	c = 1;
}
$ cat 2.c
int c;
$ riscv64-unknown-linux-gnu-gcc 1.c 2.c 
/opt/riscv64/lib/gcc/riscv64-unknown-linux-gnu/10.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccZuN9Jq.o:(.sbss+0x0): multiple definition of `c'; /tmp/ccLKf1Es.o:(.sbss+0x0): first defined here
collect2: error: ld returned 1 exit status
```
I tested the same procedure with gcc for x86_64, and it works. I checked the .o files and see on x86 the var "c" would be marked as "COM", while for riscv, it is not.
```
$ gcc -c 1.c
$ readelf -s 1.o
Symbol table '.symtab' contains 11 entries:
   Num:    Value          Size Type    Bind   Vis      Ndx Name
     0: 0000000000000000     0 NOTYPE  LOCAL  DEFAULT  UND 
     1: 0000000000000000     0 FILE    LOCAL  DEFAULT  ABS 1.c
     2: 0000000000000000     0 SECTION LOCAL  DEFAULT    1 
     3: 0000000000000000     0 SECTION LOCAL  DEFAULT    3 
     4: 0000000000000000     0 SECTION LOCAL  DEFAULT    4 
     5: 0000000000000000     0 SECTION LOCAL  DEFAULT    6 
     6: 0000000000000000     0 SECTION LOCAL  DEFAULT    7 
     7: 0000000000000000     0 SECTION LOCAL  DEFAULT    8 
     8: 0000000000000000     0 SECTION LOCAL  DEFAULT    5 
     9: 0000000000000001     1 OBJECT  GLOBAL DEFAULT  COM c
    10: 0000000000000000    18 FUNC    GLOBAL DEFAULT    1 main
```
I'm using 10.1.0 branch to build my gcc for riscv.  I want to know if we have not support this or what's our plan on this？

kito-cheng, 2020-12-03:
It's not RISC-V specific changes, it's GCC 10 change default using `-fno-common` as default[1], so in theory you can using `-fcommon` to using common section again, but personally I would suggest **fix** your program instead of using `-fcommon`, especially in your case, such kink of usage seems error prone.

[1] https://gcc.gnu.org/gcc-10/porting_to.html

unicornx, 2020-12-03:
> It's not RISC-V specific changes, it's GCC 10 change default using `-fno-common` as default[1], so in theory you can using `-fcommon` to using common section again, but personally I would suggest [...]

Thank you Kito, I double-checked my x86 gcc, it's v9.3.0.  And yes, this sample code is written by me deliberately. I will close this issue.

closed 2020-12-03 by unicornx