riscvarchive/riscv-gcc #224: readelf/objdump issue

unicornx, 2020-12-13:
When I use readelf/objdump for risc-v to dump the relocation info, I found the display info inconsistent with that displayed on x86_64.
On x86_64, the offset of global_var is the start position of data to be relocated and the display format is in little-ending and consistent with the storage in memory. But on risc-v, the offset of global_var is the  start position of the instruction and the display format is convenience for reading but not the same as arrangement in memory. I want to know if it is  a bug or just design to be？ 
The tools for risc-v is v2.35; tools for x86_64 is v2.34.

source code:
```
$ cat a.c
long global_var;

int main()
{
	return global_var != 0;
}
```

for risc-v:
```
$ riscv64-unknown-linux-gnu-gcc -c a.c
$ riscv64-unknown-linux-gnu-readelf -r a.o

Relocation section '.rela.text' at offset 0x180 contains 4 entries:
  Offset          Info           Type           Sym. Value    Sym. Name + Addend
000000000006  00080000001a R_RISCV_HI20      0000000000000000 global_var + 0
000000000006  000000000033 R_RISCV_RELAX                        0
00000000000a  00080000001b R_RISCV_LO12_I    0000000000000000 global_var + 0
00000000000a  000000000033 R_RISCV_RELAX                        0
$ riscv64-unknown-linux-gnu-objdump -rd a.o

a.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <main>:
   0:	1141                	addi	sp,sp,-16
   2:	e422                	sd	s0,8(sp)
   4:	0800                	addi	s0,sp,16
   6:	000007b7          	lui	a5,0x0
			6: R_RISCV_HI20	global_var
			6: R_RISCV_RELAX	*ABS*
   a:	0007b783          	ld	a5,0(a5) # 0 <main>
			a: R_RISCV_LO12_I	global_var
			a: R_RISCV_RELAX	*ABS*
   e:	00f037b3          	snez	a5,a5
  12:	0ff7f793          	andi	a5,a5,255
  16:	2781                	sext.w	a5,a5
  18:	853e                	mv	a0,a5
  1a:	6422                	ld	s0,8(sp)
  1c:	0141                	addi	sp,sp,16
  1e:	8082                	ret
```

for x86_64
```
$ gcc -c a.c
$ readelf -r a.o

Relocation section '.rela.text' at offset 0x200 contains 1 entry:
  Offset          Info           Type           Sym. Value    Sym. Name + Addend
00000000000b  000900000002 R_X86_64_PC32     0000000000000008 global_var - 4

Relocation section '.rela.eh_frame' at offset 0x218 contains 1 entry:
  Offset          Info           Type           Sym. Value    Sym. Name + Addend
000000000020  000200000002 R_X86_64_PC32     0000000000000000 .text + 0
$ objdump -rd a.o

a.o:     file format elf64-x86-64

Disassembly of section .text:

0000000000000000 <main>:
   0:	f3 0f 1e fa          	endbr64 
   4:	55                   	push   %rbp
   5:	48 89 e5             	mov    %rsp,%rbp
   8:	48 8b 05 00 00 00 00 	mov    0x0(%rip),%rax        # f <main+0xf>
			b: R_X86_64_PC32	global_var-0x4
   f:	48 85 c0             	test   %rax,%rax
  12:	0f 95 c0             	setne  %al
  15:	0f b6 c0             	movzbl %al,%eax
  18:	5d                   	pop    %rbp
  19:	c3                   	retq   

```

jim-wilson, 2020-12-13:
In both cases, the offset of the relocation is the address of the instruction to modify.  But RISC-V and x86 have very different instruction formats.  RISC-V instructions are always 2 or 4 bytes, and are always treated as a 2 or 4 group of bytes.  x86 has variable length instructions that are treted as a stream of bytes.  So they display differently, as x86 instructions are a list of bytes, but RISC-V instructions are 16-bits or 32-bits of info.  Since the x86 has a list of bytes, it sometimes has relocatable fields inside an instruction.  In this case, we have a 7 byte instruction, where the first 3 bytes are opcode and the last 4 bytes are data, so the relocation points at the fourth byte of the instruction.  In the RISC-V case, we have a 4-byte instruction, and the relocatable field is inside that four bytes, so we point at the first byte of the instruction.  On RISC-V, a relocation will always point at the first byte of the instruction.  On x86, sometimes it points into the middle of an instruction.

unicornx, 2020-12-14:
> In both cases, the offset of the relocation is the address of the instruction to modify. But RISC-V and x86 have very different instruction formats. RISC-V instructions are always 2 or 4 bytes, and [...]

Thank you, Jim. So for definition of field "Offset", it depends on the arch under it, right? I used to think it has unified definition as what x86 does.

jim-wilson, 2020-12-14:
Offset always has the same meaning.  But whether it points to the start of an instruction or into the middle of an instruction depends on the target's instruction formats.

unicornx, 2020-12-15:
Thanks, Jim. I will close this issue.

closed 2020-12-15 by unicornx