riscvarchive/riscv-gcc #225: [Question] about relocation types for riscv64

unicornx, 2020-12-14:
Hello,
I am comparing relocation types between x86_64 and riscv64. Regarding  `R_X86_64_GLOB_DAT` and `R_X86_64_TLSDESC`, what's the corresponding items in riscv64?
Thanks in adv.

jim-wilson, 2020-12-14:
Shrug.  I don't know x86_64 relocs very well.  But our psabi document is here
    https://github.com/riscv/riscv-elf-psabi-doc
There is probably not a one to one matching of relocs as the architecture and ABI are different.

kuanlinchentw, 2020-12-15:
Hi,
As far as I know, R_X86_64_GLOB_DAT is used to create a GOT entry to save a global address.
The familiar items in riscv64 is R_RISCV_64.

R_X86_64_TLSDESC is used to save TLS symbol descriptor in x86 -mtls-dialect=gnu2.
RISCV has not support gnu2 dialect yet. Right?

jim-wilson, 2020-12-15:
I don't know anything about tls dialiects.  But I see only aarch64 and x86 support the -mtls-dialect= option.

unicornx, 2020-12-16:
@jim-wilson @kuanlinchentw Thank you all, I will close this issue.

closed 2020-12-16 by unicornx