riscvarchive/riscv-gcc #227: [Question] how to add PHDR into shared object file?

unicornx, 2020-12-15:
Hello,

When I'm checking the program header table from the linker of android, I see it contains a PHDR. But when I use gcc (of course risc-v) to create a shared object, I can not see the PHDR. Do you know how to insert a PHDR into shared object file? What option should I use when calling ld，thanks in adv.

```
$ prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-readelf -l out/soong/.intermediates/bionic/linker/linker/android_arm_armv8-a_core/unstripped/linker

Elf file type is DYN (Shared object file)
Entry point 0x317d8
There are 10 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  PHDR           0x000034 0x00000034 0x00000034 0x00140 0x00140 R   0x4
  LOAD           0x000000 0x00000000 0x00000000 0x1ceac 0x1ceac R   0x1000
  LOAD           0x01d000 0x0001d000 0x0001d000 0x96ba8 0x96ba8 R E 0x1000
  LOAD           0x0b4000 0x000b4000 0x000b4000 0x046a0 0x0d5ec RW  0x1000
  DYNAMIC        0x0b81e8 0x000b81e8 0x000b81e8 0x00070 0x00070 RW  0x4
  GNU_RELRO      0x0b5000 0x000b5000 0x000b5000 0x036a0 0x04000 R   0x1
  GNU_EH_FRAME   0x01c2d4 0x0001c2d4 0x0001c2d4 0x002f4 0x002f4 R   0x4
  GNU_STACK      0x000000 0x00000000 0x00000000 0x00000 0x00000 RW  0
  NOTE           0x000174 0x00000174 0x00000174 0x00020 0x00020 R   0x4
  EXIDX          0x000194 0x00000194 0x00000194 0x03ce8 0x03ce8 R   0x4

 Section to Segment mapping:
  Segment Sections...
   00
   01     .note.gnu.build-id .ARM.exidx .dynsym .gnu.hash .dynstr .relr.dyn .rodata .ARM.extab .eh_frame_hdr .eh_frame
   02     .text
   03     .data .data.rel.ro .init_array .dynamic .got .bss
   04     .dynamic
   05     .data.rel.ro .init_array .dynamic .got
   06     .eh_frame_hdr
   07
   08     .note.gnu.build-id
   09     .ARM.exidx
```

kuanlinchentw, 2020-12-21:
Hi,

I think you can use linker script to add PHDRs.
Please see [3.8 PHDRS Command](https://sourceware.org/binutils/docs/ld/PHDRS.html).

unicornx, 2020-12-22:
> Hi,
> 
> I think you can use linker script to add PHDRs.
> Please see [3.8 PHDRS Command](https://sourceware.org/binutils/docs/ld/PHDRS.html).

Thanks. I will have a try.  

And I will close this issue due to I have got the answer.

closed 2020-12-22 by unicornx