riscvarchive/riscv-gcc #228: How to hack register assign in GCC for optimization

zhongjuzhe, 2020-12-17:
Here, there is a code like this:
void foo(void *addr, int32x16_t re_index)
{
  int32x16x2_t a; 
  int32x16x2_t b;
  b.val[0] = a.val[0] + b.val[1];
  b.val[1] = a.val[1] + b.val[0];
  vsstcb2_v_i32(addr, re_index, b);
}
assembler:
   10126:       0002012b                vadd.vv v2,v0,v4,mtrues
   1012a:       00058187                vld.v   v3,0(a1)
   1012e:       0022822b                vadd.vv v4,v2,v5,mtrues
   10132:       9a20002b                vmv.v.v v0,v2
   10136:       9a4000ab                vmv.v.v v1,v4
   1013a:       90356027                vsstcb2.v       v0,(a0),v3,mtrues

int32x16x2_t is a datatype I built in GCC for vector extension in my company, consist of 2 int32x16_t components(regno number is consecutive).
vsstcb2_v_i32 is an operation to store a variable which is int32x16x2_t  datatype into memory, here I find that there are two more redundant move instructions to move the vector regster and make vsstcb2 variable register consecutive.  Can I have a choice to assign 2 members of datatype int32x16x2_t in consecutive regno, so that I can remove these redundant move instructions?

kito-cheng, 2020-12-17:
Hmmm, seems like it's compile by in-house GCC, it's hard to giving something useful advice without detail info or source code.

zhongjuzhe, 2020-12-18:
> Hmmm, seems like it's compiled by in-house GCC, it's hard to giving something useful advice without detail info or source code.
Thanks for you reply, this is an example, you can see the vadd result is v2 and v4, and I want to store them into memory,in consecutive. But I think if these two vadd instructions dest reg is v0 and v1, it does not need redudant move instructions. Can I have a choice to optimize this code?

closed 2022-02-05 by zhongjuzhe