riscvarchive/riscv-gcc #229: Huge generated files fail to compile on Arm 32-bit

ilg-ul, 2020-12-24:
I'm working on compiling the 10.1 toolchain available on your 2020.08.0 release, but something major changed, and the generated files are so large, that I can no longer compile them on a Raspberry Pi, with 8 GB of RAM, when running the 32-bit version of the OS. With the 64-bit OS it passes the build.

Please take a look at the following:

```
ilg@wks gcc % wc -l insn-*.c
    3382 insn-attrtab.c
   11819 insn-automata.c
     994 insn-dfatab.c
 2540403 insn-emit.c
     128 insn-enums.c
   44683 insn-extract.c
     918 insn-latencytab.c
    4385 insn-modes.c
    6495 insn-opinit.c
 2933995 insn-output.c
      43 insn-peep.c
    1157 insn-preds.c
  490499 insn-recog.c
 6038901 total
ilg@wks gcc % 
```

As you can see, `insn-output.c` is almost 3 million lines of code! 

In the previous release, using GCC 8.3, I had no problems to compile the toolchain, even on a 4 GB system.

Any idea why these files grew so large?

ilg-ul, 2020-12-24:
For completeness, the build fails while generating files with `gcc/../move-if-change`:

```
/bin/bash /Host/home/ilg/Work/riscv-none-embed-gcc-10.1.0-1.1/linux-arm/sources/riscv-gcc-10.1.0-1.1/gcc/../move-if-change tmp-fixinc_list fixinc_list
echo timestamp > s-fixinc_list
virtual memory exhausted: Cannot allocate memory
make[2]: *** [Makefile:1117: insn-recog.o] Error 1
rm gfdl.pod gcc.pod gcov-dump.pod gcov-tool.pod fsf-funding.pod gpl.pod cpp.pod gcov.pod lto-dump.pod
make[2]: Leaving directory '/Host/home/ilg/Work/riscv-none-embed-gcc-10.1.0-1.1/linux-arm/build/gcc-10.1.0-final/gcc'
make[1]: *** [Makefile:4388: all-gcc] Error 2
make[1]: Leaving directory '/Host/home/ilg/Work/riscv-none-embed-gcc-10.1.0-1.1/linux-arm/build/gcc-10.1.0-final'
make: *** [Makefile:954: all] Error 2
```

This process takes like forever, tens of minutes on a Raspberry, and slowly increases the memory until the 4 GB limit is reached.

jim-wilson, 2020-12-25:
Is that a toolchain with vector support?  RVV has ~70K instructions if you count all of the type and vector length variations and so yes a toolchain with vector support ends up being much larger.  I don't have the 2020.08 (freedom-u-sdk? freedom-tools?) toolchain handy, but I have a copy of the riscv-gnu-toolchain rvv-instrinsics branch and I see
```
gamma17:2051$ wc -l insn-output.c
2934990 insn-output.c
```
which looks roughtly the same.

If you don't want vector support, you can work around this by deleting or commenting out (;) the Mask(VECTOR) line in gcc/config/riscv/riscv.opt and then adding a "#define TARGET_VECTOR 0" line to riscv.h in the same dir.  That works for me.  I now have a 15530 line insn-output.c file.  Completing the gcc build, I also had to add a "#define MASK_VECTOR 0" line to riscv.h and ifdef out some setvtype stuff in riscv_expand_builtin_insn in riscv-builtins.c which isn't protected by TARGET_VECTOR.  I just did a gcc build.  If there is library support for vectors then you may need additional changes to disable the vector support in the libraries.

kito-cheng, 2020-12-25:
It seems like cause by vector intrinsic, freedom-tools 2020.08 included that, and it's known issue need to resolve, but that's low priority for now.

ilg-ul, 2020-12-25:
> RVV has ~70K instructions

Really? And ~70K still counts as 'reduced' instruction set?

Is RVV really used? I mean in general and in the Freedom Studio examples in particular.

I'd prefer to disable it in my binary distribution, to also make it available for 32-bit platforms.

kito-cheng, 2020-12-25:
> Really? And ~70K still counts as 'reduced' instruction set?

GCC need combination a vector instruction with different type to implement that, so there isn't really ~70 different instruction there in the ISA level.

e.g. a vector load instruction will expand to {i8, i16, i32, i64, f16, f32, f64} x {m1, m2, m4, m8} = 28 different instruction in GCC, just for a basic vector load.

ilg-ul, 2020-12-27:
I was thinking about this, and I'm not sure what would be the less unfortunate choice, to disable vector support in my binaries and be able to provide them for all platforms, or to keep vectoring enabled and my release in sync with your release but betray the 32-bit Arm users.

It depends what your plans are regarding the vector extensions. If you plan to upstream them and make them part of the default RISC-V GCC distribution, or at least keep them in all subsequent SiFive binary distributions if not yet upstreamed, then I'll probably better keep them.

Anyway, could you check the procedure used to generate those huge files and possibly add a more elaborate logic to split them into multiple smaller files, such that the build will also pass in memory restricted environments? I would target files about ten times smaller, that will not only fit but also compile faster, since now those huge files may take tens of minutes to process on a reasonably fast machine.

aswaterman, 2020-12-28:
@ilg-ul FWIW, Spike disables vector support for 32-bit hosts (though not for the same reason), so there is already some precedent here.

kito-cheng, 2020-12-28:
Eventually vector extension will upstream, my thought upstream that after GCC 11 release, and I I would prefer fix this issue before upstream, since it's also an issue other than 32-bit platform, the build time become very long, that's really the burden during development cycle.

ilg-ul, 2020-12-28:
>  would prefer fix this issue

Great! But did you check possible solutions? I saw some discussions in bugzilla, even a ticket asking for this split, but it was not considered, so I'm afraid you have to fix it yourself.

jim-wilson, 2020-12-30:
SiFive has announced IP cores with vector support, but there is no asic you can buy, and no free fpga implementation you can download.  Andes and Alibaba have also announced cores.  There may be Alibaba cores available eary in 2021, but they implement an earlier draft and aren't compatible with our toolchain support.   The european processor initiaive (EPI) has announced a core.  And other companies/organizations are likely working on cores too.  But I don't know of any free fpga implementations or asics that you can buy.  So currently the vector support is not very useful to the general public.

Kito mentioned that each instruction has 28 variants, but if you add signed/unsigned integer, then that adds another 4*4=16 making it 44 total.  The signed/unsigned distinction shows up at the intrinsics level, but not the gcc md file pattern level.  It still contributes to the size though, as we need code to handle all of the intrinsics.

The current focus is on LLVM vector support, which will upstream before the GCC support.  The GCC support needs a lot of work and there is no one scheduled to work on it.  I think it will be a while before it can be added to mainline upstream, but I would like to get it on a vector branch in the FSF GCC tree.

ilg-ul, 2020-12-31:
Thank you, Jim; if I understand it between the lines, until vector support will be upstreamed, which might take some time, it would be better to disable vector support in the xPack binary distribution.

In this case, if following the SiFive distribution is no longer possible, probably it would be even better to use only the upstreamed sources, without any SiFive customisations.

ilg-ul, 2021-01-05:
Given the state of the upstreamed code, the option to use it as basis for the xPack RISC-V Embedded GCC distribution does not seem yet very realistic, so for now I decided to stick close to the SiFive release.

Without a single switch to enable/disable vector support, manually patching your branch to avoid generating the huge files is difficult and error prone.

FYI, compared to the previous 8.x release, build times increased by at least a factor of 2. 

https://xpack.github.io/blog/2021/01/05/riscv-none-embed-gcc-v10-1-0-1-1-released/

Unfortunately the Arm 32-bit binaries could not be built and are missing from this release.

For the next SiFive release it would be great to either split the files into more smaller files, or add the switch to disable vector support.