riscvarchive/riscv-gcc #231: Add a configuration option to disable/enable vector support

ilg-ul, 2020-12-31:
As discussed in #229, the latest SiFive 10.x GCC release includes vector support, but for various reasons building the sources is problematic, and for other distributions it is recommended to disable vector support.

Unfortunately this is not an easy task, and a single configuration option to do this would be very useful.

jim-wilson, 2020-12-31:
riscv-gcc doesn't have vector support.  Only some vendor toolchains have vector support currently.

There is an experimental branch rvv-intrinsics that has vector support, but only people that want vector support would build it, so it makes no sense to add an option there to disable vector support.  When vector support gets merged into mainline, it would make some sense to support this though.

ilg-ul, 2020-12-31:
Yes, the request refers to the changes required to upstream this.