riscvarchive/riscv-gcc #235: sprintf ecall with floating point on fully barebones system.

bitglitcher, 2021-01-18:
Hello, I'm been using `sprintf` to print to format strings, on a fully embedded program. The issue that I've noticed is that `sprintf` has `ecalls`, like if there was an underlying environment, but the system is fully barebones. The function works correctly until I want to format a floating-point number.  Which it does format the buffer, but it only does it with `0.0000` which is not the correct value. 

I wonder what is the meaning of the register when ecalling, because it supposed to be defined by the environment, but here there is no environment.
Is it because the function was meant to use along with an OS like Linux? I'm not sure... Maybe that's the case.

kito-cheng, 2021-01-19:
I guess it because `malloc` has invoked in sprintf? but you might double-check that? if so you might implement your own `_sbrk`.

bitglitcher, 2021-01-19:
I checked the trace output and the `ecall` instructions come from `_sbrk`

I dumped the whole binary and it seems that all these functions have to be implemented(By me). They all include `ecalls`.
```
00010238 <_lseek>:
00010288 <_read>:
000102d8 <_sbrk>:
00010374 <_write>:
```
I'm not sure how to replace these functions with my own implementation.

kito-cheng, 2021-01-19:
> I'm not sure how to replace these functions with my own implementation.
> Is there a way to tell the compiler to not link code that requires an underlying environment? That way I can know which things should be implemented by me.

Implement that and like with your program, and then linker will take your own one first, I would suggest you compile/link program with `-nostdlib` to prevent using crt or such system call stubs files from newlib, and then you need explicitly add `-lc -lgcc` to your linker options, you might got undef reference error from linker like bellowing message, that will be the function list you need to implement for your environment.

```
/scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 00000000000100b0
/scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-sbrkr.o): in function `_sbrk_r':
sbrkr.c:(.text+0x10): undefined reference to `_sbrk'
/scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-writer.o): in function `_write_r':
writer.c:(.text+0x16): undefined reference to `_write'
/scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-closer.o): in function `_close_r':
closer.c:(.text+0x10): undefined reference to `_close'
/scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-lseekr.o): in function `_lseek_r':
lseekr.c:(.text+0x16): undefined reference to `_lseek'
/scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-readr.o): in function `_read_r':
readr.c:(.text+0x16): undefined reference to `_read'
/scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fstatr.o): in function `_fstat_r':
fstatr.c:(.text+0x14): undefined reference to `_fstat'
/scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /scratch1/kitoc/riscv-gnu-workspace/rv64gc/install/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-isattyr.o): in function `_isatty_r':
isattyr.c:(.text+0x10): undefined reference to `_isatty'
collect2: error: ld returned 1 exit status
```

bitglitcher, 2021-01-19:
>  I would suggest you compile/link program with -nostdlib

What about if I want to use the runtime soft arithmetic functions? Such as:
```
__mulsi3
__modsi3
```

```
test.c:(.text+0x12ac): undefined reference to `__mulsi3'
/mnt/c/Users/camin/Documents/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccsl7BcA.o: in function `.L0 ':
test.c:(.text+0x12d4): undefined reference to `__mulsi3'
/mnt/c/Users/camin/Documents/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: test.c:(.text+0x1350): undefined reference to `__mulsi3'
/mnt/c/Users/camin/Documents/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: test.c:(.text+0x1378): undefined reference to `__mulsi3'
/mnt/c/Users/camin/Documents/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: test.c:(.text+0x1460): undefined reference to `memset'
/mnt/c/Users/camin/Documents/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: test.c:(.text+0x14e8): undefined reference to `rand'
/mnt/c/Users/camin/Documents/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: test.c:(.text+0x14fc): undefined reference to `__modsi3'
/mnt/c/Users/camin/Documents/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: test.c:(.text+0x1510): undefined reference to `rand'
/mnt/c/Users/camin/Documents/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: test.c:(.text+0x1524): undefined reference to `__modsi3'
/mnt/c/Users/camin/Documents/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: test.c:(.text+0x1538): undefined reference to `rand'
/mnt/c/Users/camin/Documents/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: test.c:(.text+0x154c): undefined reference to `__modsi3'
/mnt/c/Users/camin/Documents/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: test.c:(.text+0x1560): undefined reference to `rand'
```

kito-cheng, 2021-01-19:
> What about if I want to use the runtime soft arithmetic functions? Such as:

Add `-lc` and `-lgcc` to your LDFLAGS.

bitglitcher, 2021-01-20:
> > What about if I want to use the runtime soft arithmetic functions? Such as:
> 
> Add `-lc` and `-lgcc` to your LDFLAGS.

Great, thanks.

closed 2021-01-20 by bitglitcher