riscvarchive/riscv-gcc #239: How to add a new builtin function to GCC?

xicongye, 2021-02-03:
Hi,
I have some custom instructions based on vector registers, and I want toolchain to support them. So far,  I have been implemented them in binutils, and they works fine, but I get stuck when adding a new intrinsic for gcc. I read the sources code in directory `riscv-gcc/gcc/config/riscv`, I think(correct my if I'm wrong) I should:
1) implement the intrinsic functions in `riscv_vector.h`, and these intrinsic functions will invoke the builtin functions:
2) implement the builtin functions in `riscv-builtins.c`, maybe I should use the macro `DIRECT_BUILTIN` or `_RVV_INT_ITERATOR_ARG` to describe the new builtins, and attach them to the `riscv_builtins` variable; 
3) add a pattern in `vector.md`  to describe these instructions;

here is my questions:
1) how to add a new builtin function and how it works?
    I know I should implement the new builtins in `riscv-builtins.c`, but how to describe them, and what's the relationship between the builtins with assembly instructions, which I mean how gcc handles these builtins, and how to match to the correct assembly instructions;
2) how to add a new pattern to `vector.md`?
    `vector.md` has lots of iterators and attributes, it's too hard for me to clarify all of them. Could you give me some example or reference about this?

I'm a newbie to gcc, any reference or example is helpful.
Thanks.

kito-cheng, 2021-02-17:
Oh, seems like you are hacking vector intrinsic stuffs, they are kind of mess now, we have plan to reorg that for easier maintain, however it's low priory work to us, but we might drop a proof of concept in next few weeks.

I assume you are one of member of PLCT, let me know if you guys are interesting to cooperate on this work item :)

xicongye, 2021-03-02:
Hi @kito-cheng, thanks for your reply. 

Unfortunately, I am not one of member of PLCT. I found some useful information in https://gcc.gnu.org/onlinedocs/gccint/ last week.  Based on these docs, I tried to implement some intrinsic to gcc, and it seems that it works, though I don't know why it works.

Remind me if you have some useful information, thanks a lot!

pz9115, 2021-03-02:
Hello @kito-cheng ,  Jiawei from PLCT lab, we are very glad to offer some help, is there anything cooperate on this work.