riscvarchive/riscv-gcc #240: [RVV] Trying to fix testing failure after b38532d

higuoxing, 2021-02-04:
Hi, I noticed that a testcase introduced in b38532d574897180515def6cdc0279a18d7482ce is broken.

```c
// gcc/testsuite/g++.target/riscv/rvv-large-stack.C
/* { dg-do compile } */
/* { dg-options "-O0 -march=rv64gcv -mabi=lp64d" } */

#include <riscv_vector.h>
#include <random>

__attribute__((noinline)) void NestedFunc(vbool32_t a, vbool32_t b) {
  return;
}

__attribute__((noinline)) void Crash() {
  std::mt19937 rng;
  const vuint32m1_t v0 = vmv_v_x_u32m1(0);
  const vbool32_t mask = vmseq_vv_u32m1_b32(v0, v0);
  NestedFunc(mask, mask);
}

int main() {
  Crash();
  return 0;
}

/* { dg-final { scan-assembler "mul\tt0,t0,t2" } } */
//                                   ^-------- t0 and t2 are temporary registers in prologue/epilogue codes.
/* { dg-final { scan-assembler "li\tt2,-4096" } } */
/* { dg-final { scan-assembler "add\tt2,t2,t0" } } */
```

However, in `gcc/config/riscv/riscv.h`

```
// Line 358
#define RISCV_PROLOGUE_TEMP_REGNUM (GP_TEMP_FIRST + 1)
//                                  ^---------------- We actually use t1 and t2 as temporary registers in prologue/epilogue codes.
#define RISCV_PROLOGUE_TEMP(MODE) gen_rtx_REG (MODE, RISCV_PROLOGUE_TEMP_REGNUM)
#define RISCV_PROLOGUE_TEMP2_REGNUM (GP_TEMP_FIRST + 2)
//                                   ^-----------------
#define RISCV_PROLOGUE_TEMP2(MODE) gen_rtx_REG (MODE, RISCV_PROLOGUE_TEMP2_REGNUM)
```

Possible solution:

```diff
diff --git a/gcc/testsuite/g++.target/riscv/rvv-large-stack.C b/gcc/testsuite/g++.target/riscv/rvv-large-stack.C
index 43a086c0e16..9edab9a562d 100644
--- a/gcc/testsuite/g++.target/riscv/rvv-large-stack.C
+++ b/gcc/testsuite/g++.target/riscv/rvv-large-stack.C
@@ -20,7 +20,7 @@ int main() {
   return 0;
 }
 
-/* { dg-final { scan-assembler "mul\tt0,t0,t2" } } */
+/* { dg-final { scan-assembler "mul\tt1,t1,t2" } } */
 /* { dg-final { scan-assembler "li\tt2,-4096" } } */
-/* { dg-final { scan-assembler "add\tt2,t2,t0" } } */
-/* { dg-final { scan-assembler "addi\tt0,t2,1552" } } */
+/* { dg-final { scan-assembler "add\tt2,t2,t1" } } */
+/* { dg-final { scan-assembler "addi\tt1,t2,1552" } } */
```

kito-cheng, 2021-02-04:
@monkchiang could you take a look?

monkchiang, 2021-02-08:
@higuoxing, you are right. In this branch, the RISCV_PROLOGUE_TEMP_REGNUM is t1 register.
Sorry about that, I did not use this branch to reconfirm. Thank you.

higuoxing, 2021-02-08:
> @higuoxing, you are right. In this branch, the RISCV_PROLOGUE_TEMP_REGNUM is t1 register.
> Sorry about that, I did not use this branch to reconfirm. Thank you.

Thank you very much!

closed 2021-02-17 by higuoxing