riscvarchive/riscv-gcc #245: Which branch(es) are actively developed, or which branches are need CI now?

lazyparser, 2021-03-07:
I try to figure out which branch/branches should be added to CI.

https://github.com/riscv/riscv-gnu-toolchain/issues/851

I check the branches, and found that they were not updated  more than two months. Now sure whether I found the wrong place, or should I just go to upstream gcc for testing?

lazyparser, 2021-03-07:
Is
https://github.com/riscv/riscv-gcc/tree/riscv-gcc-10.2.0-bump
the right branch?

jim-wilson, 2021-03-07:
There are only two development branches, and both are problematic.  There is the riscv-gnu-toolchain rvv-instrinsics branch, but the focus for V work has moved to LLVM.  The GNU toolchain work is dormant and out-of-date.  There isn't much point to testing it.  We know it is incomplete and broken, and we have no plan for finishing it or fixing it.  There are riscv-binutils and riscv -gcc B extension branches but no riscv-gnu-toolchain B branch.  But the riscv-binutils port needed to be thrown away and rewritten from scratch due to a missing copyright assignment, so it is pointless to test it.  And the riscv-gcc port has problems too, it isn't actively maintained to due lack of time and people so isn't completely up-to-date with the new binutils port.  Also, the rvvintrin.h file needs to be thrown away due to a missing copyright assignment, and we will have to check to make sure the rest is OK.  Also, Ventana/VRULL claim to have a better B gcc port which they have been working on for almost a year but haven't let anyone look at yet.  We will likely have to merge the two ports together, taking the best of each, which will likely happen in the FSF tree.

So the only thing in the github.com/riscv tree worth testing is the master branch of riscv-gnu-toolchain.  That is the only thing we are claiming to support..

While there is nothing else interesting in the riscv tree to test, the FSF trees may be worth testing.  You can use git remote to pull FSF trees into a riscv-gnu-toolchain tree, and then use riscv-gnu-toolchain to test the FSF trees.  You could test FSF top of tree this way.  Though this is only a temporary problem, as the plan is to obsolete all of the riscv trees and make riscv-gnu-toolchain use FSF trees directly, and then you wouldn't need git tricks to test FSF mainline, you can just checkout the mainline trees in the submodules directly.  Likewise, once we have experimental branches for draft extensions in the FSF tree, they could be tested the same way.

jim-wilson, 2021-03-07:
The bump trees are temporary trees for pull requests that we sometimes forget to delete after merging in the pull request.  The riscv-gcc-10.2.0-bump tree is noit interesting.  The only interesting tree here is the one that riscv-gnu-toolchain uses.

kito-cheng, 2021-03-08:
> Is https://github.com/riscv/riscv-gcc/tree/riscv-gcc-10.2.0-bump the right branch?

I guess this branch is I was working on backport patches to riscv-gcc-10.2.0 branch, but interrupted by another works so it never getting merged.

And most work I would prefer working on trunk directly.

For rvv works, it's become kind of my side project now, I maintain that on my spare free time.

So back to the topic, I think there is only 3 branches are still activity:
- riscv-gcc-10.2.0: stable one
- riscv-gcc-10.2.0-rvb: for RVB
- riscv-gcc-10.1-rvv-dev: for RVV intrinsic

And this branch is kind of temporally used for testing big-endian:
 - riscv-gcc-be

lazyparser, 2021-03-08:
@kito-cheng @jim-wilson Thanks. Based on the information we have, I'll stop testing rvv-intrinsic and sync-after-git branches.

> > Is https://github.com/riscv/riscv-gcc/tree/riscv-gcc-10.2.0-bump the right branch?
> 
> I guess this branch is I was working on backport patches to riscv-gcc-10.2.0 branch, but interrupted by another works so it never getting merged.

Thanks. I've stopped tracking this branch.

> And most work I would prefer working on trunk directly.
> 
> For rvv works, it's become kind of my side project now, I maintain that on my spare free time.
> 
> So back to the topic, I think there is only 3 branches are still activity:
> 
>     * riscv-gcc-10.2.0: stable one

This is the default branch. So it has been tracked by ci.rvperf.org on ubuntu1804 and ubuntu2004.

>     * riscv-gcc-10.2.0-rvb: for RVB

I can add a job for this branch if helps.
 
>     * riscv-gcc-10.1-rvv-dev: for RVV intrinsic

I can add a job for this branch if helps.

> And this branch is kind of temporally used for testing big-endian:
> 
>     * riscv-gcc-be

I can add a job for this branch if helps.

closed 2021-03-08 by lazyparser

kito-cheng, 2021-03-08:
>> * riscv-gcc-10.2.0-rvb: for RVB
>
> I can add a job for this branch if helps.

I guess it need to spend some effort on that, maybe I can share what the situation on RVB toolchain with you in next sync meeting.

>> * riscv-gcc-10.1-rvv-dev: for RVV intrinsic
>
> I can add a job for this branch if helps.

It seems worth to add one.

>> And this branch is kind of temporally used for testing big-endian:
>> * riscv-gcc-be
>
> I can add a job for this branch if helps.

This can ignore at this moment, I expect (and hope) this could merge into GCC 11, and then we don't this branch here.

here still some regression on rv32be, Marcus Comstedt is working on this.

lazyparser, 2021-03-10:
Hi @jim-wilson

FYI I've set up a branch for tacking the mainline of FSF GCC/Bintuils/GDB/glibc/newlib.

https://ci.rvperf.org/blue/organizations/jenkins/fsf-gnu-toolchain-master-all-ubuntu1804/detail/fsf-gnu-toolchain-master-all-ubuntu1804/1/pipeline/

The raw log could be found at:
https://ci.rvperf.org/job/fsf-gnu-toolchain-master-all-ubuntu1804/1/console
and
https://ci.rvperf.org/job/fsf-gnu-toolchain-master-all-ubuntu1804/1/consoleText
(very long log, very big file)

A few failures now. Better than my expectation.

the tracking branch:
https://github.com/plctlab/riscv-gnu-toolchain/tree/fsf-trunk-daily-build

CI scripts are at:
https://github.com/plctlab/riscv-gnu-toolchain/blob/fsf-trunk-daily-build/jenkinsfile.ubuntu1804
https://github.com/plctlab/riscv-gnu-toolchain/blob/fsf-trunk-daily-build/jenkinsfile.ubuntu2004

If there are other branches or combinations of FSF repos interesting to you, feel free to drop a comment. I'd add them later.

lazyparser, 2021-03-10:
> > > * riscv-gcc-10.2.0-rvb: for RVB
> > 
> > 
> > I can add a job for this branch if helps.
> 
> I guess it need to spend some effort on that, maybe I can share what the situation on RVB toolchain with you in next sync meeting.

Great! I'll put this item on the agenda of next RISC-V GNU Sync.

> 
> > > * riscv-gcc-10.1-rvv-dev: for RVV intrinsic
> > 
> > 
> > I can add a job for this branch if helps.
> 
> It seems worth to add one.

https://ci.rvperf.org/job/riscv-gnu-toolchain-rvv-ubuntu1804/