riscvarchive/riscv-gcc #248: What is the condition of lw or ld when accessing the APB register?

HyunJe-Lee, 2021-03-19:
Hello.

I wrote the following to access the APB register in bit units.

```
typedef unsigned int UINT;

#define APB_REG_BASE 0x45600000

#define	_am2(uon,base,addr)		((volatile uon*)(UINT*)(base+(addr)))->a
#define	_bm2(uon,base,addr,mne)	(((volatile uon*)(UINT*)(base+(addr)))->mne)

#define	_regs_					typedef	union{UINT a; struct{
#define	_rege_					};}

#define BF2_1(a)			a;
#define BF2_2(a,...)		BF2_1(__VA_ARGS__);a;
#define BF2_3(a,...)		BF2_2(__VA_ARGS__);a;

_regs_ BF2_1(UINT MARK:32)                          _rege_ _APB_0X0000;
_regs_ BF2_3(UINT YEAR:18, UINT MONTH:7, UINT DAY:7)  _rege_ _APB_0X0004;
_regs_ BF2_3(UINT HOUR:18, UINT MIN:7, UINT SEC:7)  _rege_ _APB_0X0008;

#define APB_MARK    _bm2(_APB_0X0000,APB_REG_BASE,0x0000,MARK) // MARK[31:0]
#define APB_YEAR    _bm2(_APB_0X0004,APB_REG_BASE,0x0004,YEAR) // YEAR[31:14]
#define APB_MONTH   _bm2(_APB_0X0004,APB_REG_BASE,0x0004,MONTH) // MONTH[13:7]
#define APB_DAY     _bm2(_APB_0X0004,APB_REG_BASE,0x0004,DAY) // DAY[8:0]
#define APB_HOUR    _bm2(_APB_0X0008,APB_REG_BASE,0x0008,HOUR) // HOUR[31:14]
#define APB_MIN     _bm2(_APB_0X0008,APB_REG_BASE,0x0008,MIN) // MIN[13:7]
#define APB_SEC     _bm2(_APB_0X0008,APB_REG_BASE,0x0008,SEC) // SEC[8:0]

UINT test_abpreg(void)
{
	UINT k = 0;
	if (APB_MARK == 1) {
		k |= 10;
	}
	if (APB_DAY == 1) {
		k |= 10;
	}
	if (APB_SEC == 1) {
		k |= 10;
	}
	return k;
}
```
```
test_abpreg:
        li      a5,1163919360
        lw      a5,0(a5)                  <===== ...
        li      a4,1
        li      a0,0
        sext.w  a5,a5
        bne     a5,a4,.L2
        li      a0,10
.L2:
        li      a5,1163919360
        lw      a5,4(a5)                  <===== ...
        li      a4,1
        andi    a5,a5,127
        bne     a5,a4,.L3
        li      a0,10
.L3:
        li      a5,1163919360
        ld      a5,8(a5)                  <===== ld?
        li      a4,1
        andi    a5,a5,127
        beq     a5,a4,.L7
        ret
.L7:
        li      a0,10
        ret
```
code: https://godbolt.org/z/zoqxTr

Why is it compiled with 'lw' or 'ld' depending on the address or bit area being accessed?
Found in both RISC-V rv64gc gcc 10.2.0 and 8.2.0.

I want to access all of them with 'lw'.

kito-cheng, 2021-03-19:
More reduced case: 
```c
typedef unsigned int UINT;
typedef union{UINT a; struct{ UINT MARK:32; };} _APB_0X0000;
typedef union{UINT a; struct{ UINT SEC:7;;UINT MIN:7;;UINT HOUR:18; };} _APB_0X0008;

UINT test_abpreg(void)
{
 UINT k = 0;
 if ((((volatile _APB_0X0000*)(UINT*)(0x45600000 +(0x0000)))->MARK) == 1) {
  k |= 10;
 }
    if ((((volatile _APB_0X0008*)(UINT*)(0x45600000 +(0x0008)))->SEC) == 1) {
  k |= 10;
 }
 return k;
}
```

jim-wilson, 2021-03-19:
It is a compiler optimization.  When accessing a bit-field, we try to load the largest size that we can from that address, so that we can load multiple bit-fields with a single load instruction, instead of emitting a separate load instruction for each bit-field.  In the first case, you are loading a 32-bit value not a bit-field, so you get a lw.  In the second case, you are loading a bit-field, but it is a 4-byte aligned address, so the largest load we can use is lw.  In the third case, you are loading a bit-field, and it is 8-byte aligned, so the largest load we can use is ld.

Note that this isn't a RISC-V issue.  You will see the same behavior for other targets.  An x86_64 gcc does the same thing for instance.

In the gcc backend, the decision on whether to widen a bit-field depends on whether it is volatile.  If volatile, then it checks the narrow_volatile_bitfield hook, which is false for all targets other than the rx.  If not volatile, then it depends on !SLOW_BYTE_ACCESS, which is also false for RISC-V.  So we will always widen bit-field accesses.

If you really need a 4-byte access, then I think you need to rewrite your code to avoid a bit-field, and use a mask and shift instead.

HyunJe-Lee, 2021-03-21:
Thank you for reply.

I'll use 'mask' and 'shift', like you said.
Thank you!

closed 2021-03-21 by HyunJe-Lee