riscvarchive/riscv-gcc #249: Do we need a new branch for k-ext ?

WuSiYu, 2021-03-30:
I'm going to submit a PR for my gcc work for k-ext, it's based on the `riscv-gcc-10.2.0` branch.
I wonder what branch that I should PR to, the `riscv-gcc-10.2.0` branch here or a new branch for k-ext (something like `riscv-gcc-10.2.0-rvk`). @kito-cheng

kito-cheng, 2021-03-30:
We'll migrate the development to FSF repo included all draft extensions after GCC 11 release, but I am happy to review the patch before that happen, you can sent PR to `riscv-gcc-experiment` branch, so that we can review the patch.

Just double confirm for FSF copyright issue, are you PLCT folks? or did you has signed FSF copyright assignment?

@lazyparser

WuSiYu, 2021-03-30:
> Just double confirm for FSF copyright issue, are you PLCT folks? or did you has signed FSF copyright assignment?

Yes, I'm from PLCT, and the commit is using the iscas's mail address, I think the copyright is fine.

Thanks and I will submit the PR later.

closed 2021-03-30 by WuSiYu

lazyparser, 2021-03-30:
yes. Siyu Wu and Sinan Lin are PLCT's interns :-)