riscvarchive/riscv-gcc #252: Breakpoint problem in GDB

zhongjuzhe, 2021-04-07:
8000137e:       02f76263                bltu    a4,a5,800013a2 <bkfir32x32i_process+0xa4>
80001382:       e285                    bnez    a3,800013a2 <bkfir32x32i_process+0xa4>
80001384:       fbfaf793                andi    a5,s5,-65

This is the assembly instructions that I consecutive set 3 breakpoints. Breakpoint 1 at pc 0x8000137e,  length = 2. Breakpoint 2 at pc   0x80001380,length = 4. Break 3 at pc 0x80001382, length = 2. It seems that the lenght and pc offset is not matched.  The GDB version I use is riscv-binutils-2.35.  I don't really understand the way GDB send breakpoint.  Is this a BUG in GDB? Thank you!

jim-wilson, 2021-04-07:
RISC-V has 2-byte and 4-byte breakpoints.  We can't use a 2-byte breakpoint unless we know that the target supports C.  There is no way for gdb to sure of what the target hardware can do.  We can't write a 4-byte breakpoint over top of a 2-byte instruction, as that might clobber the first 2 bytes of the next instruction.  So gdb will attempt to read target memory to see what the instruction is.  If we can read and decode the instruction, then we emit a breakpoint of the same size.  if we can't read target memory, then we emit a 2-byte breakpoint if the address is unaligned, since only a C instruction can be at an unaligned address, and some targets don't support unaligned writes (i.e. can't write a 4-byte breakpoint to a 2-byte aligned address).  And we emit a 4-byte breakpoint if the address is aligned.  See the riscv_breakpoint_kind_from_pc function in gdb/riscv-tdep.c.  It sounds like maybe gdb can't read your target memory.  Also, you are trying to set a breakpoint at a non-instruction address which is asking for trouble, there is no instruction at 0x80001380.

What I described is the default behavior.  There is a command to change it.  "show riscv use-compressed-breakpoints" will show you the current value.  "set riscv use-compressed-breakpoints auto/off/on will set it to the chosen value.  auto is the default.  We emit a 2-byte breakpoint only if we can prove it is safe, otherwise a 4-byte breakpoint.  If on, then we always emit a 2-byte breakpoint.  If off, then we always emit a 4-byte breakpoint.

zhongjuzhe, 2021-05-27:
> in gdb/riscv-tdep.c. It sounds like maybe gdb can't read your target memory. Also, you are trying to set a breakpoint at a non-instruction address which is asking for trouble, there is no instruction at 0x80001380.
> 
> What I described is the default behavior. There is a comm

Thanks!

closed 2022-02-05 by zhongjuzhe