riscvarchive/riscv-gcc #254: Test fail about vector extension of branch 'riscv-gcc-10.1-rvv-dev'

rjiejie, 2021-04-13:
e.g.

from gcc/testsuite/gcc.target/riscv/rvv/rvv_load-stack.c:

#include <riscv_vector.h>
void foo(int32_t *c){
   int32_t arr1[20] = {1,2, 33};
   int32_t arr2[20] = {0};
   vint32m1_t va, vb, vc;
   va = vle32_v_i32m1((int32_t*)&arr1);
   vb = vle32_v_i32m1((int32_t*)&arr2);
   vc = va + vb;
   vse32_v_i32m1(c, vc);
}

The prototype of 'vle32_v_i32m1' is:

vle##SEW##ff_v_i##SEW##m##LMUL (const int##SEW##_t *a, word_type *new_vl, word_type vl)

kito-cheng, 2021-04-14:
It's known issue, we don't have update most testsuite to reflect API change yet, it's would be great if you could help.

rjiejie, 2021-04-15:
Okay, I will make a patch for this issue :)

closed 2021-04-21 by rjiejie