riscvarchive/riscv-gcc #259: RISC-V Vector Intrinsics types missing casts between same size vectors and groups of vectors

tony-cole, 2021-04-27:
Re: riscv32-unknown-elf-gcc (GCC) 10.1.0 (intrinsic-patch)

I can't find a way to cast  **vfloat32m4_t**  to  **vfloat32m2x2_t**  and visa-versa. Please let me know if this is currently possible or not.

I'm trying to:

    load vfloat32m4_t
    extract the lower vfloat32m2_t and use that for processing
    slide down the vfloat32m4_t
    load the upper vfloat32m2_t of the vfloat32m4_t
    repeat, etc

I found the tuple and segment load/store intrinsics which almost do what I want, but I can't cast a **vfloat32m4_t** to **vfloat32m2x2_t** and visa-versa, GCC gives an error if I just use the normal C casting method.

I need GCC to accept explicit C casts between equivalent sized vector types, or add more intrinsics to do the same, something like:

    vfloat32m2x2_t   vcast_f32m4_f32m2x2 (vfloat32m4_t   src);
    vfloat32m4_t     vcast_f32m2x2_f32m4 (vfloat32m2x2_t src);

and of course all the other type combinations that have the same register (LMUL) sizes. 

For me the explicit C casting method is preferable (implicit casting should still generate an error as this is very useful for catching mistakes). 

There will be many other uses, for instance casting vuint8m8_t to vint8m8_t or to any of the other types (e.g. vuint32m8_t).

jim-wilson, 2021-04-28:
You may have better luck asking in the rvv-intrinsics-doc github tree.  There is no active development on the gcc rvv support at present.

GCC in general does allow you to cast from one vector type to another if they are the same size, but this is more complicated for RVV as all vector types have variable size, so no two vector types can be assumed to have the same size.  I don't know if the operation you need exists and/or is supported by gcc.

tony-cole, 2021-04-29:
This also has been posted on: https://github.com/riscv/rvv-intrinsic-doc/issues/83