riscvarchive/riscv-gcc #261: A wrong email address is used in a commit of a previous PR

WuSiYu, 2021-05-12:
Sorry to disturb. When I am checking the git log today, I notice that [this commit](https://github.com/riscv/riscv-gcc/commit/8b64ef1f1543eaa228a9dc5cc1ec1654627b3f05.patch) in my previous PR (#250), is using a wrong email address.

It should be the `siyu@isrc.iscas.ac.cn` which has the copyright assignment, but may due to the wrong git config of my another PC, in that commit is my personal email address `wu.siyu@hotmail.com`.

Is it will be a copyright issue?  Should we revert it and re-commit it?

Sorry again for my careless.

kito-cheng, 2021-05-12:
It's OK to me, we could fix that when summit patch to upstream.

@jim-wilson did you have concern about that?

jim-wilson, 2021-05-12:
This is fine.  We can correct it later as you suggested.

WuSiYu, 2021-05-13:
OK I know, thanks for this.

closed 2021-05-13 by WuSiYu