riscvarchive/riscv-gcc #263: RVV Intrinsic Error During Compiling with rv64imafcv

lzwButton, 2021-05-17:
Hello.
I have failed to compile my project with `https://github.com/riscv/riscv-gcc/tree/riscv-gcc-10.1-rvv-dev` branch. The `arch` is `rv64imafcv` . The reference function is as below. (It may not look quite useful, because I deleted most part of my function.)

```
#include "riscv_vector.h"

void test_func_f64(float64_t * pDst)
{
    size_t l;
    vfloat64m1_t v_temp;
    l = vsetvl_e64m1(1);
    v_temp = vfsub_vv_f64m1(v_temp, v_temp, l);
    pDst[0] = vfmv_f_s_f64m1_f64(v_temp);
}
/* Here is the main function which is not important*/
```

Then I got the error message as below:
```
error: unable to generate reloads for:
   10 | }
      | ^
(insn 45 44 46 2 (parallel [
            (set (mem:DF (reg/v/f:DI 138 [ pDst ]) [1 *pDst_3(D)+0 S8 A64])
                (unspec:DF [
                        (vec_select:DF (reg:VNx2DF 136 [ _6 ])
                            (parallel [
                                    (const_int 0 [0])
                                ]))
                        (reg:SI 66 vl)
                    ] UNSPEC_USEVL))
            (use (reg:VNx2DI 67 vtype))
        ]) "/(Here is the toolchain directory)/10.1.0/include/riscv_vector.h":826:1 11610 {vec_extract_fextvnx2df_nosetvl}
     (expr_list:REG_DEAD (reg:VNx2DF 136 [ _6 ])
        (expr_list:REG_DEAD (reg:VNx2DI 67 vtype)
            (expr_list:REG_DEAD (reg:SI 66 vl)
                (expr_list:REG_EQUIV (mem:DF (reg/v/f:DI 138 [ pDst ]) [1 *pDst_3(D)+0 S8 A64])
                    (nil))))))
during RTL pass: reload
```

As far as I tested, this error only happens with float point. The same code can be compiled successfully with `rv64imafdcv`. Because `ELEN ≥ max(XLEN,FLEN)`, I assume that even with `FLEN=32` the `XLEN=64` and it should support `vfloat64_t` and `f64 sized instructions`.

jim-wilson, 2021-05-19:
I got multiple error with your testcase, tried to fix them, and then gave up.  The snippet can't be compiled.

The error indicates that the compiler tried to emit a scalar D instruction as part of register allocation, and failed, because the scalar D support wasn't enabled.  That will likely be a hard problem for a compiler to work around.

I don't think that you can assume that FP double support is supported by the vector extension just because XLEN is 64-bits.  This is likely implementation defined.  There may be targets feeding vector instructions through the scalar ALU for instance, and if you don't have D support then vector double instructions can't work.  Note that ELEN can still be 64 and not have FP double support if 64-bit integers are supported by the vector extension.  Some of these values really should be compiler options, but the gcc vector port doesn't support that.

Note in general that the gcc vector support is dormant, and there is plan to restart work on it.  You are better off using LLVM for vector work instead.

lzwButton, 2021-05-19:
Thank you for your comment.

closed 2021-05-19 by lzwButton