riscvarchive/riscv-gcc #266: RISCV LLVM toolchain buildup problem

zhongjuzhe, 2021-05-27:
Recently,I try to build up LLVM RISCV toolchain base on LLVM 12.0 down load from llvm.org. I got link error.
[image]

I wonder whether RISCV LLVM toolchain has its own ld.lld. Or it should rely on GCC linker? 
Can anyone help me? Thanks

kito-cheng, 2021-05-27:
The error message show that you are missing compiler-rt library, clang/llvm could using libgcc instead, either you build compiler-rt or indicate GNU toolchian path by `--gcc-toolchain=` could resolve the issue.

zhongjuzhe, 2021-05-27:
> The error message show that you are missing compiler-rt library, clang/llvm could using libgcc instead, either you build compiler-rt or indicate GNU toolchian path by `--gcc-toolchain=` could resolve the issue.

Thanks for you reply.  I search many resources on google. I try many method. But I always encounter linker fail. 
First I try this method: https://github.com/sifive/riscv-llvm. This method should use GCC compile finally.
Second I try this method:https://github.com/lowRISC/riscv-llvm. It doesn't work.
Finall I found this method:https://github.com/andestech/riscv-llvm-toolchain. This is the toolchain you build 4 years ago, It works now. This is the only method I tried which can work.  It seems that it should use GCC toolchain. Can I build up LLVM toolchain totally isolated from GCC toolchain? Can you help me with that? I was struggling in this problem so much.Thank you so much!

zhongjuzhe, 2021-05-27:
> The error message show that you are missing compiler-rt library, clang/llvm could using libgcc instead, either you build compiler-rt or indicate GNU toolchian path by `--gcc-toolchain=` could resolve the issue.

Does compiler-rt support RISCV now? Or still we should rely on libgcc for RISCV target on LLVM?

kito-cheng, 2021-05-27:
> Does compiler-rt support RISCV now? Or still we should rely on libgcc for RISCV target on LLVM?

Yes, compiler-rt support RISC-V, but you need assemble the toolchain right to let clang/LLVM find that.

if you are not family with building toolchain you could try to using https://github.com/andestech/riscv-llvm-toolchain but without updated source, I think it should work fine with current LLVM.

zhongjuzhe, 2021-05-27:
> Yes, compiler-rt support RISC-V, but you need assemble the toolchain right to let clang/LLVM find that.
>
> if you are not family with building toolchain you could try to using [...]

Thank you! I have tried your toolchain environment and replace the clang, llvm with current LLVM. It works now.  But still I have several questions. 
1. I read the makefile in your toolchain, I dont see compilation of compiler-rt. And also I see it still use riscv-gnu-toolchain libgcc. How can I replace libgcc with compiler-rt?
2. It still need riscv-gnu-toolchain which is GCC toolchain. I am confused why it still needs riscv-gnu-toolchain and which module in riscv-gnu-toolchain is needed for clang+llvm RISCV toolchain.
Thanks~

cmuellner, 2021-05-27:
> if you are not family with building toolchain you could try to using https://github.com/andestech/riscv-llvm-toolchain but without updated source, I think it should work fine with current LLVM.

Is the `llvm` branch in https://github.com/riscv/riscv-gnu-toolchain intended to provide this or does that have another purpose?

kito-cheng, 2021-05-27:
> It still need riscv-gnu-toolchain which is GCC toolchain. I am confused why it still needs riscv-gnu-toolchain and which module in riscv-gnu-toolchain is needed for clang+llvm RISCV toolchain.
Thanks~

I didn't spend too much time on integrating compiler-rt, although it's port by me...

[1] https://reviews.llvm.org/D42958

kito-cheng, 2021-05-27:
> Is the llvm branch in https://github.com/riscv/riscv-gnu-toolchain intended to provide this or does that have another purpose?

Yeah, it's kind of experimental stuffs, but I thought it's weird to merge it into master branch of riscv-***gnu***-toolchain, so I just keep that in separated branch there.

zhongjuzhe, 2021-05-27:
> I didn't spend too much time on integrating compiler-rt, although it's port by me...
>
> [1] https://reviews.llvm.org/D42958

Can CLANG+LLVM toolchain totally be isolated from riscv-gnu-toolchain? Or it still rely on some stuff of riscv-gnu-toolchain?
And can you give me some hints on integrating compiler-rt?  Thank you~

kito-cheng, 2021-05-27:
Ooops, I suddenly found a fact is the issue is created at `riscv-gcc` rather than `riscv-gnu-toolchain` lol...

kito-cheng, 2021-05-27:
> Can CLANG+LLVM toolchain totally be isolated from riscv-gnu-toolchain? Or it still rely on some stuff of riscv-gnu-toolchain?
And can you give me some hints on integrating compiler-rt? Thank you~

I believe the answer is yes, but `lld` didn't support linker relaxation yet, so you might got worse performance and bigger code size.

And I would suggest you could send mail to llvm mailing list for how to assemble a pure LLVM toolchain.

zhongjuzhe, 2021-05-27:
> I believe the answer is yes, but `lld` didn't support linker relaxation yet, so you might got worse performance and bigger code size.
>
> And I would suggest you could send mail to llvm mailing list [...]

Ok, Thank you!
I got two more questions:
1.Can I use ld of riscv-gnu-toolchain and other components like llvm,clang,compiler-rt,...etc which are not in riscv-gnu-toolchain   and assemble them to avoid this relaxation problem?
2.Can you tell how to set searching path of clang compiler because it always search the path on riscv-gnu-toolchain so it fails to search compiler-rt.

kito-cheng, 2021-06-03:
I gonna close this issue since it's wrong place to discuss the building issue for LLVM toolchain.

closed 2021-06-03 by kito-cheng