riscvarchive/riscv-gcc #270: gnu toolchain build fail

zhongjuzhe, 2021-06-03:
WARNING: 'automake-1.15' is missing on your system.
         You should only need it if you modified 'Makefile.am' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'automake' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
gmake[3]: *** [Makefile:1174: ../../../riscv-gcc/isl/Makefile.in] Error 1
gmake[2]: *** [Makefile:6240: all-isl] Error 2
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [Makefile:445: stamps/build-gcc-newlib-stage1] Error 2
make: *** [Makefile:97: gcc] Error 2

When I use another server to build gnu toolchain, it failed. Is the automake version necessary to be 1.15? my automake version is 1.14.Thanks!

kito-cheng, 2021-06-03:
Those error message is not enough to know what happen, but it seems something wrong during building `isl` package inside gcc.

kito-cheng, 2021-06-03:
I would suggest those building issue go https://github.com/riscv/riscv-gnu-toolchain instead, riscv-gcc repo are used for reporting about gcc developing issue.

jim-wilson, 2021-06-03:
There is a configure time check for gmp, mpc, and mpfr libraries.  If missing, we will run gcc's contrib/download_prerequisites to download them.  But this script actually downloads four libraries, those three plus isl.  isl is optional.  You can prevent the script from running if you install the gmp, mpc, and mpfr libraries on the build machine.  The prerequisites section of the riscv-gnu-toolchain README has info on how to install these packages.  Please look at that.

We could avoid the unnecessary isl download by passing --no-isl when running the download_prerequisites script.

The underlying problem here is that some file timestamps in the isl dir are wrong, but they do appear to be correct in the downloaded tar file, so it isn't clear why they are wrong in your build.

closed 2022-02-05 by zhongjuzhe