riscvarchive/riscv-gcc #274: How to avoid Double-precision floating-point extension instructions.

Kendidi, 2021-06-17:
Hi,

Not sure if this is the right place to post for this question.  If not, please kindly direct me.  Thanks a lot in advance!

The RV32 chip I am working on supports `Single-precision floating-point extension` but not `Double-precision floating-point extension`. 

The compiled code uses instructions **fcvt.d.s** and **fcvt.s.d** which I think are only supported in `Double-precision floating-point extension`.  Is there a compiler option can be used to stop the compiler from using those instructions?  Please advise.

Command I am using now:

    riscv32-unknown-elf-gcc -g -ffreestanding -O0 -nostartfiles -Triscv32_00-virt.ld -o  riscv32_fw.elf startup.s main.c uart.c

jim-wilson, 2021-06-17:
-march=rv32imafc -mabi=ilp32f for instance will emit code for a target with F but not D support, and will pass float but not double args in FP registers.

While this works for the compiler, it doesn't work for pre-compiled libraries, like the C library and libgcc.  So you either need a multilibbed compiler that includes rv32imafc/ilp32f as one of the multilibs, or you need to configure and build a compiler using --with-arch=rv32imafc --with-abi=ilp32f to get a compiler that uses the right arch/abi by default.

jim-wilson, 2021-06-17:
There is an old blog series by Palmer Dabbelt that talks about how the RISC-V toolchain works.
https://www.sifive.com/blog/all-aboard-part-1-compiler-args

Kendidi, 2021-06-17:
I see.  So it appears to be a toolchain build issue.  Thanks a lot for your advice, @jim-wilson !!  I will try to rebuild toolchain with `--with-abi=ilp32f` and see how it goes.

Kendidi, 2021-06-17:
Encounter the following while rebuilding the toolchain.  Any idea?

> In file included from /opt/riscv32/sysroot/usr/include/pthread.h:27,
>                  from ./gthr-default.h:35,
>                  from ../../.././riscv-gcc/libgcc/gthr.h:148,
>                  from ../../.././riscv-gcc/libgcc/libgcov-interface.c:27:
> /opt/riscv32/sysroot/usr/include/bits/setjmp.h:35:3: error: #error unsupported FLEN
>    35 | # error unsupported FLEN
>       |   ^~~~~
> Makefile:924: recipe for target '_gcov_dump.o' failed
> make[3]: *** [_gcov_dump.o] Error 1
> ......
> ......

jim-wilson, 2021-06-17:
Are you trying to build a linux compiler?  The linux toolchain only supports soft-float and double-float.  Your original message mentioned riscv32-unknown-elf-gcc which is an embedded compiler, and can support single-float.

Kendidi, 2021-06-17:
I am trying to rebuild `riscv-gnu-toolchain`.  The toolchain is being used to cross-compile firmware to be run on a RV32 platform.

jim-wilson, 2021-06-17:
it looks like you are trying to build a riscv-linux compiler.  That won't work.  Build a riscv-elf compiler instead.  If you used "--enable-linux" or "make linux" that is wrong.  That will give a riscv-linux compiler which is not what you want.

You can always try downloading a compiler if you are having trouble building one.  The SiFive web site has toolchains that can be downloaded on the software page.

Kendidi, 2021-06-17:
You are right, @jim-wilson ! I have rebuilt the toolchain with the following and now instructions `fcvt.d.s` and `fcvt.s.d` are not used anyway. 

    ./configure --prefix=/opt/riscv32 --with-arch=rv32imfc --with-abi=ilp32f
    sudo make

Floating point arithmetic seems ok at runtime.  But now it failed when I tried to print the result to a buffer with _sprintf()_.

    778	    sprintf(str, "x = %f\n\r", x);
    (gdb) s
    __extendsfdf2 (a=1.90576591e-43) at ../../.././riscv-gcc/libgcc/soft-fp/extendsfdf2.c:44
    44	  FP_INIT_EXCEPTIONS;
    (gdb) 
    45	  FP_UNPACK_RAW_S (A, a);
    (gdb) 
    47	  FP_EXTEND (D, S, 2, 1, R, A);
    (gdb) 
    51	  FP_PACK_RAW_D (r, R);
    (gdb) 
    52	  FP_HANDLE_EXCEPTIONS;
    (gdb) 
    54	  return r;
    (gdb)

Kendidi, 2021-06-17:
I wonder is there an option for choosing `soft-FP` or `hard-FP`?

jim-wilson, 2021-06-17:
arch and abi choose whether this is soft-fp or hard-fp.  If you choose an arch and abi with no float, then you get soft-float.  If you choose an arch and abi with double float, then you get hard-float.  You can also choose things in between, e.g. you can use an arch with float and an ABI without float, or an arch and abi with only single-float, etc.  Since we have more than two choices, a soft-float/hard-float option makes no sense.

sprintf should work.  You didn't say what exactly failed.  You can't have a float arg to sprintf, as it will always be promoted to double, and hence require a soft-float convesion from float to double in your case.  But it should still work.  I don't know off hand if newlib provides a way to convert a float to a string without a conversion to double first.

Kendidi, 2021-06-17:
It froze after `return r;`.  Hard to pressed `CTRL-C' to get out.

    54	  return r;
    (gdb) 
    ^C
    Program received signal SIGINT, Interrupt.
    0x1a000000 in ?? ()
    (gdb)

jim-wilson, 2021-06-17:
Shrug.  Maybe you don't have enough stack space and something got overridden?  Or maybe your sprintf buffer is not large enough?  I know that sprintf works when compiling code to run on a simulator.

Kendidi, 2021-06-17:
I see.

The following appears to indicate soft-FP is used.  Am I correct?

    __extendsfdf2 (a=1.90576591e-43) at ../../.././riscv-gcc/libgcc/soft-fp/extendsfdf2.c:44

If so, is it because of the following combination I used to build the toolchain?  If yes, I wonder how to change to hard-float (single-precision)?

    --with-arch=rv32imfc --with-abi=ilp32f

jim-wilson, 2021-06-17:
It isn't soft-float.  It is single-float in hardware and double-float in software.  You can't pass a single float to sprintf, it must always be promoted to double.  If you don't want the single float to double conversion, then don't use sprintf.  Though you may not have much choice unless you want to try to write your own routine for converting single float numbers to a string.

Kendidi, 2021-06-18:
How to promote single float to double before invoking sprintf()?

I have tried the following but still failed.

    sprintf(str, "x = %f\n\r", (double)x);

jim-wilson, 2021-06-18:
It doesn't matter whether you explicitly or implicitly promote float to double.  You get the exact same code either way, which is soft-float code, as you don't have double float instructions.  If you want to avoid the soft-float code, then you have to avoid using double.  Which is harder than it sounds, because the ISO C standard says that when you pass a float to a stdarg routine it is automatically promoted to double.  So you can't call any stdarg routines like sprintf.

Maybe what you want is a toolchain where double is 32-bit same as float, and hence float to double conversions are a no-op?  We don't support that.

I doubt that your code is failing due to a compiler problem.  I think something more fundamental is wrong, like not enough stack space, or a sprintf destination buffer that isn't big enough to hold the string that is being generated.  Maybe try snprintf instead?

Kendidi, 2021-06-18:
Stack space should be OK.

We are using a sprintf() routine from a file tailored for embedded systems.  Mostly like that's where the issue occurs.  Thanks @jim-wilson for your great help!!  Appreciate it,